mrpt-ros-pkg/mrpt_navigation #16: " cannot find -lmrpt-base-dbg" problem

772358053, 2015-12-11:
I build mrpt-1.3.2 from source, and I want to build mrpt_navigation from source.
I use these commands:
```
$ cd catkin_ws/src
$ git clone https://github.com/mrpt-ros-pkg/mrpt_navigation.git
$ cd ..
$ catkin_make
```
But when I run ``catkin_make`` , I got 2 errors like this:
```
siat@siat-ThinkPad:~/catkin_ws$ catkin_make
Base path: /home/siat/catkin_ws
Source space: /home/siat/catkin_ws/src
Build space: /home/siat/catkin_ws/build
Devel space: /home/siat/catkin_ws/devel
Install space: /home/siat/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/siat/catkin_ws/build"
####
####
#### Running command: "make -j4 -l4" in "/home/siat/catkin_ws/build"
####
[  0%] [  0%] Built target std_msgs_generate_messages_py
Built target geometry_msgs_generate_messages_py
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] Built target geometry_msgs_generate_messages_lisp
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target geometry_msgs_generate_messages_cpp
[  1%] [  2%] Built target talker
Built target test1
Linking CXX shared library /home/siat/catkin_ws/devel/lib/libmrpt_bridge.so
/usr/bin/ld: cannot find -lmrpt-base-dbg
/usr/bin/ld: cannot find -lmrpt-opengl-dbg
/usr/bin/ld: cannot find -lmrpt-obs-dbg
/usr/bin/ld: cannot find -lmrpt-maps-dbg
/usr/bin/ld: cannot find -lmrpt-vision-dbg
/usr/bin/ld: cannot find -lmrpt-tfest-dbg
/usr/bin/ld: cannot find -lmrpt-slam-dbg
collect2: error: ld returned 1 exit status
make[2]: *** [/home/siat/catkin_ws/devel/lib/libmrpt_bridge.so] 错误 1
make[1]: *** [mrpt_navigation/mrpt_bridge/CMakeFiles/mrpt_bridge.dir/all] 错误 2
make[1]: *** 正在等待未完成的任务....
[  2%] [  2%] Built target _mrpt_msgs_generate_messages_check_deps_ObservationRangeBearing
Built target _mrpt_msgs_generate_messages_check_deps_SingleRangeBearingObservation
Linking CXX shared library /home/siat/catkin_ws/devel/lib/libmrpt_localization_core.so
/usr/bin/ld: cannot find -lmrpt-base-dbg
/usr/bin/ld: cannot find -lmrpt-opengl-dbg
/usr/bin/ld: cannot find -lmrpt-gui-dbg
/usr/bin/ld: cannot find -lmrpt-obs-dbg
/usr/bin/ld: cannot find -lmrpt-vision-dbg
/usr/bin/ld: cannot find -lmrpt-tfest-dbg
/usr/bin/ld: cannot find -lmrpt-maps-dbg
/usr/bin/ld: cannot find -lmrpt-slam-dbg
collect2: error: ld returned 1 exit status
make[2]: *** [/home/siat/catkin_ws/devel/lib/libmrpt_localization_core.so] 错误 1
make[1]: *** [mrpt_navigation/mrpt_localization/CMakeFiles/mrpt_localization_core.dir/all] 错误 2
make: *** [all] 错误 2
Invoking "make -j4 -l4" failed

```
Could anyone help me?

jlblancoc, 2015-12-11:
The `-dbg` suffix in `-lmrpt-base-dbg` and alike means that your are bulding ROS nodes in Debug configuration... is that intentional?? 

If you are sure that is what you want, just recompile MRPT from sources in Debug mode, setting CMAKE_BUILD_TYPE=Debug in CMake. I recommend building MRPT in Release and Debug in two separate directories, and passing MRPT_DIR to catkin / CMake accordingly depending on your build type. 

Hope it helps.

closed 2015-12-11 by jlblancoc

772358053, 2015-12-11:
@jlblancoc Thank you for your help. I used Debug mode before, and when I change it to Release mode by ``$ catkin_make CMAKE_BUILD_TYPE=Release``, it compiled okay!