mrpt-ros-pkg/mrpt_navigation #21: How to publish pose topic just like /amcl_pose in AMCL?

Megacephalo, 2016-03-30:
My colleague presented to me MRPT localization which appears to render better pose estimate than AMCL. However, the mrpt_localization_node does not yet publish similar topic as amcl_pose (geometry_msgs/PoseWithCovarianceStamped)  in AMCL. So, it becomes apparent to think about such possibility of implementing a similar publisher for the mentioned node.
I found that mrpt_localization.cpp does publish the mean pose using meanPose.asString().c_str(); but at present I have little knowledge on how to convert the MRPT data type to the one using geometry_msgs/PoseWithCovarianceStamped. Is it possible to convert in that way? Or it is better to directly calculate the mean pose from the published PosePointCloud topic? 

I am really looking forward to use mrpt_localization_node in better localizing Turtlebot !! Thank you in advance.

jlblancoc, 2016-03-30:
Hi @Megacephalo ! 

If you feel like doing it, it would be great if you could fork & submit a PR with this feature, knowing the following MRPT-specific part: 

        mrpt::math::CMatrixDouble33 cov; // cov for x,y,phi (meters, radians)
        mrpt::poses::CPose2D mean;
        pdf_.getCovarianceAndMean(cov,mean);
        // Access the mean like: mean.x(), mean.y(), mean.phi()
        // Access the cov like usual in Eigen: cov(0,0), ...cov(2,2)

Then, save the mean and covariance in a ROS msg and publish it as usual.

Megacephalo, 2016-04-07:
Hi, @jlblancoc ,
I did add an additional method to publish pose, but I fear that the same feature can just be implemented within publishTF (). Thus I dare not to push a request to you. What I did is to publish the topic /mrpt_pose which publishes the data type geometry_msgs/PoseWithCovarianceStamped, but I am not sure if it is okay to use for I have not tried it yet. I am sorry that I returned to my own project this time and have scarce time to implement such functionality. Should I still push the request for you? Thank you again for making MRPT possible for ROS !

jlblancoc, 2016-04-07:
Definitively, I'll be better than nothing! :-) 

So, go on for the PR... we will have time in the future for any improvement, if needed.

closed 2016-04-07 by jlblancoc