mrpt-ros-pkg/mrpt_navigation #25: librawlog_play.so: cannot open shared object file

galou, 2016-08-03:
When launching `roslaunch mrpt_localization demo_ro.launch` I get the error `librawlog_play.so: cannot open shared object file`.

There must be a dependency missing. Which package provides `librawlog_play.so` under Ubuntu 14.04 with Indigo?

jlblancoc, 2016-08-03:
Hi, 

Have you installed mrpt-ros packages using `apt-get` or manually compiling them?

galou, 2016-08-04:
Hi,

I installed the packages:
libmrpt-base1.0
libmrpt-detectors1.0
libmrpt-dev
libmrpt-gui1.0
libmrpt-hmtslam1.0
libmrpt-hwdrivers1.0
libmrpt-kinematics1.0
libmrpt-maps1.0
libmrpt-obs1.0
libmrpt-opengl1.0
libmrpt-reactivenav1.0
libmrpt-scanmatching1.0
libmrpt-slam1.0
libmrpt-topography1.0
libmrpt-vision1.0
mrpt-apps
ros-indigo-mrpt-bridge
ros-indigo-mrpt-local-obstacles
ros-indigo-mrpt-localization
ros-indigo-mrpt-msgs
ros-indigo-mrpt-rawlog
ros-indigo-mrpt-reactivenav2d

jlblancoc, 2016-08-04:
That seems right.

What's your output from:

        echo $LD_LIBRARY_PATH

?

It maybe that you didn't call ros' setup.bash?

galou, 2016-08-04:
Thanks again for the quick reply!

`echo $LD_LIBRARY_PATH` returns `/usr/lib/x86_64-linux-gnu:/home/gael/ros_indigo_ws/devel/lib:/opt/ros/indigo/lib:/opt/ros/indigo/lib/x86_64-linux-gnu::/usr/local/lib64`.

I don't think the problem is related to $LD_LIBRARY_PATH because the file is not present on my system (`locate librawlog_play.so` returns false, after `updatedb`).

jlblancoc, 2016-08-05:
I have reviewed the packages in the ROS servers and indeed this seems a bug in the CMake install scripts. Thanks for notifying! 

hopefully this is fixed now. The packages are to be uploaded to the servers, it may take 1-2 days to be available for download. Your fix should come by doing a `sudo apt-get update && sudo apt-get upgrade`, but as I said I can't tell how many hours / days will it take. 

Cheers.

closed 2016-08-05 by jlblancoc

galou, 2016-08-05:
Thanks a lot! I'll wait the few days, it's Friday anyway!

jlblancoc, 2016-08-10:
The builds on the farms are OK, but it seems to me that the .deb packages are not updated on a daily basis in the public `apt-get` servers, so it may take longer than expected to see the updated packages available for download...