mrpt-ros-pkg/mrpt_navigation #39: Build error in ros-indigo

sai5555, 2016-10-24:
My os : Ubuntu 14.04 and ros-indigo

Today I update mrpt_navigation code then rebuild that provide error

> /usr/include/mrpt/base/include/mrpt/math/CMatrixTemplate.h: In constructor ‘mrpt::math::CMatrixTemplate<T>::CMatrixTemplate(std::size_t, std::size_t, V (&)[N])’:
/usr/include/mrpt/base/include/mrpt/utils/mrpt_macros.h:257:83: error: there are no arguments to ‘static_assert’ that depend on a template parameter, so a declaration of ‘static_assert’ must be available [-fpermissive]
  #define MRPT_COMPILE_TIME_ASSERT(expression) static_assert(expression,#expression);
                                                                                   ^
/usr/include/mrpt/base/include/mrpt/math/CMatrixTemplate.h:160:5: note: in expansion of macro ‘MRPT_COMPILE_TIME_ASSERT’
     MRPT_COMPILE_TIME_ASSERT(N!=0)
     ^
/usr/include/mrpt/base/include/mrpt/utils/mrpt_macros.h:257:83: note: (if you use ‘-fpermissive’, G++ will accept your code, but allowing the use of an undeclared name is deprecated)
  #define MRPT_COMPILE_TIME_ASSERT(expression) static_assert(expression,#expression);
                                                                                   ^
/usr/include/mrpt/base/include/mrpt/math/CMatrixTemplate.h:160:5: note: in expansion of macro ‘MRPT_COMPILE_TIME_ASSERT’
     MRPT_COMPILE_TIME_ASSERT(N!=0)

how I fix it ? Thanks.

jlblancoc, 2016-10-24:
Is it possible that you are using MRPT from the PPA repository? If yes, I could imagine what is the problem, but I want to know first for sure what version of MRPT do you have...

sai5555, 2016-10-24:
Thanks your reply.
How I  check mrpt version ?

The following are using dpkg -L grep mrpt

> libmrpt-base1.0
libmrpt-base1.5
libmrpt-detectors1.5
libmrpt-dev
libmrpt-graphslam1.5
libmrpt-gui1.0
libmrpt-gui1.5
libmrpt-hmtslam1.5
libmrpt-hwdrivers1.5
libmrpt-kinematics1.5
libmrpt-maps1.0
libmrpt-maps1.5
libmrpt-nav1.5
libmrpt-obs1.0
libmrpt-obs1.5
libmrpt-opengl1.0
libmrpt-opengl1.5
libmrpt-reactivenav1.0
libmrpt-scanmatching1.0
libmrpt-slam1.0
libmrpt-slam1.5
libmrpt-tfest1.5
libmrpt-topography1.5
libmrpt-vision1.0
libmrpt-vision1.5
mrpt-apps
mrpt-common
ros-indigo-mrpt-bridge
ros-indigo-mrpt-icp-slam-2d
ros-indigo-mrpt-local-obstacles
ros-indigo-mrpt-localization
ros-indigo-mrpt-map
ros-indigo-mrpt-msgs
ros-indigo-mrpt-navigation
ros-indigo-mrpt-rawlog
ros-indigo-mrpt-reactivenav2d
ros-indigo-mrpt-tutorials

jlblancoc, 2016-10-25:
I realized this is the same error we're trying to solve in this other thread: https://github.com/mrpt-ros-pkg/mrpt_slam/pull/13

In theory, the latest PPA packages that are building right now should fix this by doing a `sudo apt-get update && sudo apt-get upgrade` in a couple of hours, so I'll close this. Feel free to follow the other thread and/or reopen if it's not solved.
Cheers

sai5555, 2016-10-26:
> In theory, the latest PPA packages that are building right now should fix this by doing a sudo apt-get update && sudo apt-get upgrade in a couple of hours, so I'll close this. Feel free to follow the other thread and/or reopen if it's not solved.

After I updated PPA packages , it's ok for mrpt_navigation and mrpt_SLAM
Very thank you.

closed 2016-10-26 by jlblancoc