mrpt-ros-pkg/mrpt_navigation #41: Using "sudo apt-get install ros-indigo-mrpt-ekf-slam-2d" doesn't install the package

andrestoga, 2016-10-27:
Hi,

I used the command "sudo apt-get install ros-indigo-mrpt-ekf-slam-2d" to install the package but only appears one folder and one file: "/opt/ros/indigo/share/mrpt_ekf_slam_2d" which are: cmake  package.xml. I checked the repository https://github.com/mrpt-ros-pkg/mrpt_slam/tree/master/mrpt_ekf_slam_2d and it has more files.

Best,

jlblancoc, 2016-10-27:
@Logrus : I think the CMake `install` command is missing from all the slam packages (!!). It went unnoticed since we all test the packages by building from sources, I guess...
Could you please take a look at this? I won't be able to do it within a few days...
As a reference, please see for example the `install()` CMake commands in `mrpt_navigation` packages:
https://github.com/mrpt-ros-pkg/mrpt_navigation/blob/master/mrpt_localization/CMakeLists.txt#L170

@andrestoga thanks for reporting! In the meanwhile, you could install MRPT from PPA, and build the ROS packages from sources. I'm afraid that's the quickest solution...

Logrus, 2016-10-31:
@jlblancoc sure, I'll take a look at it asap.

Logrus, 2016-10-31:
https://github.com/mrpt-ros-pkg/mrpt_slam/commit/54e1f52d9a20baa8477a6e6f4f338dbd6bbb5fe9 hopefully fixes it.

jlblancoc, 2016-11-01:
Thanks! 👍 
Will make a new ROS release soon so these changes get to the final users...

closed 2016-11-01 by jlblancoc