mrpt-ros-pkg/mrpt_navigation #44: Missing mrpt-tfest lib; is it compiled by default on mrpt?

corot, 2016-11-10:
I'm trying to build debian packages for both mrpt and mrpt_navigation. Mrpt went fine (apart from issue https://github.com/MRPT/mrpt/pull/367), but mrpt_bridge fails not finding  -lmrpt-tfest:
```
Linking CXX shared library devel/lib/libmrpt_bridge.so
/usr/bin/cmake -E cmake_link_script CMakeFiles/mrpt_bridge.dir/link.txt --verbose=1
/usr/bin/x86_64-linux-gnu-g++  -fPIC -g -O2 -fstack-protector --param=ssp-buffer-size=4 -Wformat -Werror=format-security -DNDEBUG -D_FORTIFY_SOURCE=2   -Wno-deprecated -Wno-long-long -Wno-variadic-macros    -shared -Wl,-soname,libmrpt_bridge.so -o devel/lib/libmrpt_bridge.so CMakeFiles/mrpt_bridge.dir/src/pose.cpp.o CMakeFiles/mrpt_bridge.dir/src/point_cloud.cpp.o CMakeFiles/mrpt_bridge.dir/src/point_cloud2.cpp.o CMakeFiles/mrpt_bridge.dir/src/laser_scan.cpp.o CMakeFiles/mrpt_bridge.dir/src/map.cpp.o CMakeFiles/mrpt_bridge.dir/src/beacon.cpp.o CMakeFiles/mrpt_bridge.dir/src/landmark.cpp.o  -L/opt/ros/indigo/lib /opt/ros/indigo/lib/libtf.so /opt/ros/indigo/lib/libtf2_ros.so /opt/ros/indigo/lib/libactionlib.so /opt/ros/indigo/lib/libmessage_filters.so /opt/ros/indigo/lib/libroscpp.so -lboost_signals -lboost_filesystem /opt/ros/indigo/lib/libxmlrpcpp.so /opt/ros/indigo/lib/libtf2.so /opt/ros/indigo/lib/librosconsole.so /opt/ros/indigo/lib/librosconsole_log4cxx.so /opt/ros/indigo/lib/librosconsole_backend_interface.so -llog4cxx -lboost_regex /opt/ros/indigo/lib/libroscpp_serialization.so /opt/ros/indigo/lib/librostime.so -lboost_date_time /opt/ros/indigo/lib/libcpp_common.so -lboost_system -lboost_thread -lpthread -lconsole_bridge -lmrpt-base -lmrpt-opengl -lmrpt-obs -lmrpt-maps -lmrpt-vision -lmrpt-tfest -lmrpt-slam -Wl,-rpath,/opt/ros/indigo/lib: 
/usr/bin/ld: cannot find -lmrpt-tfest
collect2: error: ld returned 1 exit status
```
So...  this library is not compiled by default on mrpt? How should I enable building it?
Thanks!

jlblancoc, 2016-11-10:
Yes, it is compiled!!??! :-S

Have you considered using the packages built in [our PPA](https://launchpad.net/~joseluisblancoc/+archive/ubuntu/mrpt)?? They are updated daily. At least you could use them for testing the build of the ROS packages against MRPT...

corot, 2016-11-10:
Actually, I would love to do so! But... I tried to compile mrpt_navigation using libmrpt-dev instead of ros-indigo-mrpt and got some problems:
```
/usr/bin/ld: cannot find -lmrpt-scanmatching
....
catkin_ws/devel/.private/mrpt_rawlog/lib/librawlog_play.so: undefined reference to `mrpt::slam::CActionRobotMovement2D::TMotionModelOptions::TMotionModelOptions()'
collect2: error: ld returned 1 exit status
.....
catkin_ws/src/mrpt_navigation/mrpt_reactivenav2d/src/mrpt_reactivenav2d_node.cpp:42:58: fatal error: mrpt/nav/reactive/CReactiveNavigationSystem.h: No such file or directory
 # include <mrpt/nav/reactive/CReactiveNavigationSystem.h>
```
When building against sources all these problems are gone

corot, 2016-11-11:
mmm....  I updated packages from the PPA and all worked like a charm....

jlblancoc, 2016-11-11:
Good news! :+1:

closed 2016-11-11 by jlblancoc