mrpt-ros-pkg/mrpt_navigation #46: mrpt_localization asserts as nFreeCells == 0 when using certain maps

corot, 2016-11-16:
The assert is this:
```
[/mrpt_localization_node  INFO 1479336265.697575744, 6.000000000]: InitializeFilter: 0.000m, 0.000m, 0.000rad 
terminate called after throwing an instance of 'std::logic_error'
  what():  

 =============== MRPT EXCEPTION =============
void mrpt::slam::CMonteCarloLocalization2D::resetUniformFreeSpace(mrpt::maps::COccupancyGridMap2D*, double, int, double, double, double, double, double, double), line 286:
Assert condition failed: nFreeCells
void mrpt::slam::CMonteCarloLocalization2D::resetUniformFreeSpace(mrpt::maps::COccupancyGridMap2D*, double, int, double, double, double, double, double, double), line 310:

```
located [here](https://github.com/MRPT/mrpt/blob/master/libs/slam/src/slam/CMonteCarloLocalization2D.cpp#L285)

Is in mrpt, so maybe I should open the issue there. I do here because the problem is with ROS maps used by mrpt_localization as gridmaps, so... well make sense to me handle in a ROS context.

This issue showed up out of issue #43. As @jlblancoc said there, my map has no free space:
[image]
```
image: localization.pgm
resolution: 0.025
origin: [-2.81815169334, -2.72403180122, 0.000000]
negate: 0
occupied_thresh: 0.65
free_thresh: 0.196
```
Well.... it's crowded, admittedly. But even if I paint big areas in white, mrpt_localization still complains. However, this other map seems fine for mrpt:
[image]
```
free_thresh: 0.157
image: localization.pgm
negate: 0
occupied_thresh: 0.2
origin: [0.0, 0.0, 0.0]
resolution: 0.025
```
So no idea why one map is ok and the other not. I tried using the same values in both yaml but same result, so the difference is in the bitmaps. 
I was using the same map without problem until very recently (last week, I think), so I suppose it must be something very new (I use the latets sources for both mrpt and mrpt_localization)

Thanks!

jlblancoc, 2016-11-18:
Hi, 

Well, both maps clearly have **plenty** of empty space, so you should definitively not have this assertion (!). 

How are you loading the map into the ROS node? Via a map server? I guess it must be related to the threshold used in MRPT to determine which cells are free or occupied...

corot, 2016-11-19:
Hi!
I use map server. I think I found the reason: this assertion happens if the initial position of the robot (normally 0, 0, 0) lays in unknown or obstacle cells. In the first map, I moved the center of the map by subtracting 8 and 2  from origin coordinates, so 0, 0, 0 is in the central empty area, and the problem is gone. Not fully sure, but I think the algorithm should be tolerant to a wrong initial pose, as the user will probably provide (as we do) an stored position through /initialpose topic. But this assertion kills mrpt_localization before this can be done.

corot, 2016-11-21:
:see_no_evil: :see_no_evil: :see_no_evil: :see_no_evil: :see_no_evil: :see_no_evil: :see_no_evil: :see_no_evil: 
I'm an idiot...  all comes from the fact that I initialize robot pose with zero covariance.
I didn't think about that because we always used amcl, who don't care at all about the initial pose covariance (he uses parameters to generate the initial particle distribution)
Sorry for the spamming!!!

closed 2016-11-21 by corot