mrpt-ros-pkg/mrpt_navigation #60: ‘THROW_EXCEPTION_CUSTOM_MSG1’ was not declared in this scope

andrestoga, 2017-02-16:
Hi,

Trying to compile the code I get the following error:

```
In static member function ‘static const bool mrpt_bridge::MapHdl::loadMap(mrpt::maps::CMultiMetricMap&, const mrpt::utils::CConfigFile&, const string&, const string&, bool)’:
/home/andrestoga/ros_Map_Merging/src/mrpt_navigation/mrpt_bridge/src/map.cpp:201:94: error: ‘THROW_EXCEPTION_CUSTOM_MSG1’ was not declared in this scope
             THROW_EXCEPTION_CUSTOM_MSG1("Map file has unknown extension: '%s'",mapExt.c_str());
```

Best,

jlblancoc, 2017-02-16:
Hi @andrestoga ! 

This one should have also gone after updating to the PPA, right? 

These are the nuisances of "unstable" versions, sorry!

andrestoga, 2017-02-16:
Nope, I commented it out and it works after that.

majorx234, 2017-02-21:
Hi,
have same issue when compiling with mrpt 1.5.
solved it by adding following line in mrpt_bridge/src/map.cpp:
#if MRPT_VERSION>=0x150
#define THROW_EXCEPTION_CUSTOM_MSG1(msg,param1) \ 
THROW_EXCEPTION_FMT(msg,param1)

andrestoga, 2017-02-23:
Thanks! It works!

jlblancoc, 2017-02-26:
It's now fixed in master. 
Cheers!

commit 2a16e753 (2017-02-26): Fix #60
closed 2017-02-26 by jlblancoc