mrpt-ros-pkg/mrpt_navigation #64: build error for indigo

sai5555, 2017-03-06:
Hi 
I update latest version from ppa
> sudo add-apt-repository ppa:joseluisblancoc/mrpt
> sudo apt-get update
> sudo apt-get install libmrpt-dev mrpt-apps  
And git clone from https://github.com/mrpt-ros-pkg/mrpt_navigation.git
When I build them (catkin_make) , show error message are the following. 
>  /usr/bin/ld: cannot find -lmrpt-base-dbg
> /usr/bin/ld: cannot find -lmrpt-opengl-dbg
> /usr/bin/ld: cannot find -lmrpt-gui-dbg
> /usr/bin/ld: cannot find -lmrpt-obs-dbg
> /usr/bin/ld: cannot find -lmrpt-graphs-dbg
> /usr/bin/ld: cannot find -lmrpt-vision-dbg
> /usr/bin/ld: cannot find -lmrpt-tfest-dbg
> /usr/bin/ld: cannot[ 15%]  find -lmrpt-maps-dbg
> /usr/bin/ld: cannot find -lmrpt-slam-dbg
> collect2: error: ld returned 1 exit status
> [ 15%] make[2]: *** [/home/robot/dev/catkin_ws/localizationfordynamic/devel/lib/libmrpt_localization_core.so] Error 1

jlblancoc, 2017-03-07:
Hi, 

Have you tried removing the catkin_ws subdirectories `build` and `devel` and running `catkin_make` again?

It looks like you used some IDE that set (or set it manually somewhere?) BUILD_TYPE to Debug...

sai5555, 2017-03-09:
Thanks
This my fault.
After I deleted build and devel folders , I re-built mrpt that is pass.

closed 2017-03-09 by jlblancoc