mrpt-ros-pkg/mrpt_navigation #67: Error compiling after 22 March 2017

sirsomething, 2017-04-26:
Hello!

After 22 March, more or less, I can't compile the mrpt navigation pakage anymore, as I already said to you.
I think it somehow the mrpt_reactivena2d node again.

`/home/jorge/catkin_ws/src/mrpt_navigation/mrpt_reactivenav2d/src/mrpt_reactivenav2d_node.cpp:269:23: error: cannot declare field ‘ReactiveNav2DNode::m_reactive_if’ to be of abstract type ‘ReactiveNav2DNode::MyReactiveInterface’
  MyReactiveInterface  m_reactive_if;
                       ^
/home/jorge/catkin_ws/src/mrpt_navigation/mrpt_reactivenav2d/src/mrpt_reactivenav2d_node.cpp:111:9: note:   because the following virtual functions are pure within ‘ReactiveNav2DNode::MyReactiveInterface’:
  struct MyReactiveInterface :
         ^
In file included from /usr/include/mrpt/nav/include/mrpt/nav/reactive/CAbstractNavigator.h:11:0,
                 from /usr/include/mrpt/nav/include/mrpt/nav/reactive/CWaypointsNavigator.h:11,
                 from /usr/include/mrpt/nav/include/mrpt/nav/reactive/CAbstractPTGBasedReactive.h:12,
                 from /usr/include/mrpt/nav/include/mrpt/nav/reactive/CReactiveNavigationSystem.h:12,
                 from /home/jorge/catkin_ws/src/mrpt_navigation/mrpt_reactivenav2d/src/mrpt_reactivenav2d_node.cpp:42:
/usr/include/mrpt/nav/include/mrpt/nav/reactive/CRobot2NavInterface.h:49:16: note: 	virtual bool mrpt::nav::CRobot2NavInterface::getCurrentPoseAndSpeeds(mrpt::math::TPose2D&, mrpt::math::TTwist2D&, mrpt::system::TTimeStamp&, mrpt::math::TPose2D&, std::__cxx11::string&)
   virtual bool getCurrentPoseAndSpeeds(
`

It would be nice if someone could fix it since I'm very new to ROS and I dont understand any of those errors.

Thanks!

bergercookie, 2017-04-27:
Hi @sirsomething, 

This is not a permanent solution, merely a quickfix for you to continue working... 
If you don't need to use something from the mrpt_reactivenav2d package, you can skip its compilation altogether by adding an empty file named CATKIN_IGNORE right under the package directory, that is`mrpt_reactivenav2d/CATKIN_IGNORE`

jlblancoc, 2017-04-27:
It's fixed after the few last commits: https://github.com/mrpt-ros-pkg/mrpt_navigation/commit/faad76b3a132c0a59f05b6fe36acc07ecc8cfa51

Cheers!

closed 2017-04-27 by jlblancoc