mrpt-ros-pkg/mrpt_navigation #76: roslaunch mrpt_localization demo_ro.launch error

dahlan77, 2017-06-22:
when i run with roslaunch

i got ERROR: cannot launch node of type [mrpt_localization/mrpt_localization_node]: can't locate node [mrpt_localization_node] in package [mrpt_localization]

and catkin_make is not compiled
make[2]: *** [mrpt_navigation/mrpt_localization/CMakeFiles/mrpt_localization.dir/src/mrpt_localization/mrpt_localization.cpp.o] Error 1
make[1]: *** [mrpt_navigation/mrpt_localization/CMakeFiles/mrpt_localization.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: *** [mrpt_navigation/mrpt_reactivenav2d/CMakeFiles/mrpt_reactivenav2d_node.dir/src/mrpt_reactivenav2d_node.cpp.o] Error 1
make[1]: *** [mrpt_navigation/mrpt_reactivenav2d/CMakeFiles/mrpt_reactivenav2d_node.dir/all] Error 2
make: *** [all] Error 2

so what should i do??

best regard

Iqbal A. Dahlan
Postgraduate 
Bandung Institute of Technology

jlblancoc, 2017-07-03:
Hi, 

Clearly, your problem is that the node has not been compiled... but the part of the error messages you show doesn't help determining the cause! The real error should be printed above those lines...

dahlan77, 2017-07-04:
[ 98%] Built target rawlog_record_node
In file included from /usr/include/mrpt/base/include/mrpt/utils.h:84:0,
                 from /usr/include/mrpt/base/include/mrpt/base.h:30,
                 from /home/dahlan/catkin_ws/src/mrpt_navigation/mrpt_localization/src/mrpt_localization/mrpt_localization.cpp:35:
/usr/include/mrpt/base/include/mrpt/utils/pose_utils.h:47:28: error: default template arguments may not be used in function templates without -std=c++11 or -std=gnu++11
     mrpt::poses::CPose3D& p) {
                            ^
/usr/include/mrpt/base/include/mrpt/utils/pose_utils.h: In function ‘void mrpt::utils::internal::getPoseFromString(const string&, mrpt::poses::CPose3D&)’:
/usr/include/mrpt/base/include/mrpt/utils/pose_utils.h:51:60: note: #pragma message: TODO: Is the transformation of frames needed?
    MRPT_TODO("Is the transformation of frames needed?");
                                                            ^
/home/dahlan/catkin_ws/src/mrpt_navigation/mrpt_reactivenav2d/src/mrpt_reactivenav2d_node.cpp: In member function ‘virtual bool ReactiveNav2DNode::MyReactiveInterface::getCurrentPoseAndSpeeds(mrpt::math::TPose2D&, mrpt::math::TTwist2D&, mrpt::system::TTimeStamp&, mrpt::math::TPose2D&, std::string&)’:
/home/dahlan/catkin_ws/src/mrpt_navigation/mrpt_reactivenav2d/src/mrpt_reactivenav2d_node.cpp:163:58: note: #pragma message: TODO: Retrieve current speeds from odometry
    MRPT_TODO("Retrieve current speeds from odometry");
                                                          ^
/home/dahlan/catkin_ws/src/mrpt_navigation/mrpt_reactivenav2d/src/mrpt_reactivenav2d_node.cpp: In member function ‘virtual bool ReactiveNav2DNode::MyReactiveInterface::senseObstacles(mrpt::maps::CSimplePointsMap&, mrpt::system::TTimeStamp&)’:
/home/dahlan/catkin_ws/src/mrpt_navigation/mrpt_reactivenav2d/src/mrpt_reactivenav2d_node.cpp:244:50: note: #pragma message: TODO: TODO: Check age of obstacles!
    MRPT_TODO("TODO: Check age of obstacles!");
                                                  ^
/home/dahlan/catkin_ws/src/mrpt_navigation/mrpt_reactivenav2d/src/mrpt_reactivenav2d_node.cpp: In member function ‘void ReactiveNav2DNode::navigateTo(const mrpt::math::TPose2D&)’:
/home/dahlan/catkin_ws/src/mrpt_navigation/mrpt_reactivenav2d/src/mrpt_reactivenav2d_node.cpp:379:20: error: ‘struct mrpt::nav::CAbstractNavigator::TargetInfo’ has no member named ‘x’
   navParams.target.x = target.x;
                    ^
/home/dahlan/catkin_ws/src/mrpt_navigation/mrpt_reactivenav2d/src/mrpt_reactivenav2d_node.cpp:380:20: error: ‘struct mrpt::nav::CAbstractNavigator::TargetInfo’ has no member named ‘y’
   navParams.target.y = target.y ;
                    ^
/home/dahlan/catkin_ws/src/mrpt_navigation/mrpt_reactivenav2d/src/mrpt_reactivenav2d_node.cpp:381:13: error: ‘struct mrpt::nav::CAbstractPTGBasedReactive::TNavigationParamsPTG’ has no member named ‘targetAllowedDistance’
   navParams.targetAllowedDistance = m_target_allowed_distance;
             ^
/home/dahlan/catkin_ws/src/mrpt_navigation/mrpt_reactivenav2d/src/mrpt_reactivenav2d_node.cpp:382:13: error: ‘struct mrpt::nav::CAbstractPTGBasedReactive::TNavigationParamsPTG’ has no member named ‘targetIsRelative’
   navParams.targetIsRelative = false;
             ^
make[2]: *** [mrpt_navigation/mrpt_localization/CMakeFiles/mrpt_localization.dir/src/mrpt_localization/mrpt_localization.cpp.o] Error 1
make[1]: *** [mrpt_navigation/mrpt_localization/CMakeFiles/mrpt_localization.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
^[[Amake[2]: *** [mrpt_navigation/mrpt_reactivenav2d/CMakeFiles/mrpt_reactivenav2d_node.dir/src/mrpt_reactivenav2d_node.cpp.o] Error 1
make[1]: *** [mrpt_navigation/mrpt_reactivenav2d/CMakeFiles/mrpt_reactivenav2d_node.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
always failed in 98 % .actually i have installed all the packages

dahlan77, 2017-07-04:
setting /run_id to 5f1d75da-604e-11e7-9307-2477034f7c04
process[rosout-1]: started with pid [6742]
started core service [/rosout]
process[rviz-2]: started with pid [6759]
process[rawlog_play_node-3]: started with pid [6760]
ERROR: cannot launch node of type [mrpt_localization/mrpt_localization_node]: can't locate node [mrpt_localization_node] in package [mrpt_localization]
when i ros launched

rsmallegoor, 2017-08-24:
> error: default template arguments may not be used in function templates without -std=c++11 or -std=gnu++11

jlblancoc, 2017-12-10:
No activity in a long time, and probably now fixed in last C++11 code updates to the repo, so closing it.

closed 2017-12-10 by jlblancoc