mrpt-ros-pkg/mrpt_navigation #84: CMake Error: could not find "marker_msgsConfig.cmake"

lvchaoshun, 2017-09-09:
I compiling the mrpt 1.5 from source, and then I git the mrpt_navigation ros package,
But when I run catkin_make 

Call Stack (most recent call first):
  mrpt_navigation/mrpt_bridge/CMakeLists.txt:8 (find_package)

-- Could not find the required component 'marker_msgs'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "marker_msgs" with
  any of the following names:

    marker_msgsConfig.cmake
    marker_msgs-config.cmake

  Add the installation prefix of "marker_msgs" to CMAKE_PREFIX_PATH or set
  "marker_msgs_DIR" to a directory containing one of the above files.  If
  "marker_msgs" provides a separate development package or SDK, be sure it
  has been installed.

Anyone can help me....

lvchaoshun, 2017-09-09:
My OS is ubuntu 14.04LTS, ROS indigo

maxbader, 2017-09-09:
on Ubuntu 16.04 you should be able to install ros-kinetic-marker-msgs or
the msgs from source http://wiki.ros.org/marker_msgs.
The mrpt_rawlog pkg is now able to log this msgs as bearing range or range
rawlog data.

Greetings

-- 
Dipl.-Ing. Dr.tech. Markus Bader

Institute of Computer Aided Automation
Automation Systems Group
TECHNISCHE UNIVERSITÄT WIEN
Treitlstrasse 1 | 4. Floor | E183-1 | 1040 Wien
DVR-Nummer: 0005886
Tel. +43 (0)1 58801 - 18305
Skype: markus.bader.at.work
Fax. +43 (0)1 58801 - 18391
markus.bader@tuwien.ac.at | www.auto.tuwien.ac.at |
http://info.tuwien.ac.at/mbader/

lvchaoshun, 2017-09-10:
Thanks @maxbader , It works!  
But I get a new error:
CMake Error at mrpt_navigation/mrpt_bridge/CMakeLists.txt:89 (add_dependencies):
  The dependency target "mrpt-base" of target "test_Bridge" does not exist.
CMake Error at mrpt_navigation/mrpt_bridge/CMakeLists.txt:89 (add_dependencies):
  The dependency target "mrpt-graphs" of target "test_Bridge" does not exist.
CMake Error at mrpt_navigation/mrpt_bridge/CMakeLists.txt:89 (add_dependencies):
  The dependency target "mrpt-maps" of target "test_Bridge" does not exist.
CMake Error at mrpt_navigation/mrpt_bridge/CMakeLists.txt:89 (add_dependencies):
  The dependency target "mrpt-obs" of target "test_Bridge" does not exist.
CMake Error at mrpt_navigation/mrpt_bridge/CMakeLists.txt:89 (add_dependencies):
  The dependency target "mrpt-opengl" of target "test_Bridge" does not exist.
CMake Error at mrpt_navigation/mrpt_bridge/CMakeLists.txt:89 (add_dependencies):
  The dependency target "mrpt-slam" of target "test_Bridge" does not exist.
CMake Error at mrpt_navigation/mrpt_bridge/CMakeLists.txt:89 (add_dependencies):
  The dependency target "mrpt-tfest" of target "test_Bridge" does not exist.
CMake Error at mrpt_navigation/mrpt_bridge/CMakeLists.txt:89 (add_dependencies):
  The dependency target "mrpt-vision" of target "test_Bridge" does not exist.

closed 2017-09-14 by lvchaoshun

majorx234, 2017-10-08:
get the same error
@lvchaoshun how did you fixed it?

majorx234, 2017-10-09:
solved it by myself problem is ${MRPT_LIBS} in line 89. 
removed it and problem is gone
opened an [issue](https://github.com/mrpt-ros-pkg/mrpt_navigation/issues/90)