mrpt-ros-pkg/mrpt_navigation #85: catkin_make Error

lvchaoshun, 2017-09-14:
I installed mrpt 1.5.3 from source on my Debian 9 stretch , and I execute " sudo make install ".

There is some error when i compile the mrpt-ros-pkg/mrpt_navigation package use catkin_make .

-- +++ processing catkin package: 'mrpt_localization'
-- ==> add_subdirectory(mrpt_navigation-master/mrpt_localization)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Configuring done
CMake Error at mrpt_navigation-master/mrpt_bridge/CMakeLists.txt:89 (add_dependencies):
  The dependency target "mrpt-base" of target "test_Bridge" does not exist.

CMake Error at mrpt_navigation-master/mrpt_bridge/CMakeLists.txt:89 (add_dependencies):
  The dependency target "mrpt-graphs" of target "test_Bridge" does not exist.

CMake Error at mrpt_navigation-master/mrpt_bridge/CMakeLists.txt:89 (add_dependencies):
  The dependency target "mrpt-maps" of target "test_Bridge" does not exist.

CMake Error at mrpt_navigation-master/mrpt_bridge/CMakeLists.txt:89 (add_dependencies):
  The dependency target "mrpt-obs" of target "test_Bridge" does not exist.

CMake Error at mrpt_navigation-master/mrpt_bridge/CMakeLists.txt:89 (add_dependencies):
  The dependency target "mrpt-opengl" of target "test_Bridge" does not exist.

CMake Error at mrpt_navigation-master/mrpt_bridge/CMakeLists.txt:89 (add_dependencies):
  The dependency target "mrpt-slam" of target "test_Bridge" does not exist.

CMake Error at mrpt_navigation-master/mrpt_bridge/CMakeLists.txt:89 (add_dependencies):
  The dependency target "mrpt-tfest" of target "test_Bridge" does not exist.

CMake Error at mrpt_navigation-master/mrpt_bridge/CMakeLists.txt:89 (add_dependencies):
  The dependency target "mrpt-vision" of target "test_Bridge" does not exist.

-- Generating done
-- Build files have been written to: /home/lvchaoshun/catkin_ws/build
Invoking "cmake" failed

Anyone can help me, thanks!

lvchaoshun, 2017-09-16:
Now the output is:
Scanning dependencies of target mrpt_bridge
[ 46%] Building CXX object mrpt_navigation-0.1.8/mrpt_bridge/CMakeFiles/mrpt_bridge.dir/src/pose.cpp.o
[ 47%] Building CXX object mrpt_navigation-0.1.8/mrpt_bridge/CMakeFiles/mrpt_bridge.dir/src/point_cloud.cpp.o
[ 48%] Building CXX object mrpt_navigation-0.1.8/mrpt_bridge/CMakeFiles/mrpt_bridge.dir/src/point_cloud2.cpp.o
/home/lvchaoshun/bzrobot_ws/bz_external/mrpt_navigation-0.1.8/mrpt_bridge/src/point_cloud2.cpp: In function 'bool mrpt_bridge::copy(const mrpt::maps::CSimplePointsMap&, const Header&, sensor_msgs::PointCloud2&)':
/home/lvchaoshun/bzrobot_ws/bz_external/mrpt_navigation-0.1.8/mrpt_bridge/src/point_cloud2.cpp:108:51: note: #pragma message: TODO: Implement pointcloud2 mrpt2ros
   MRPT_TODO("Implement pointcloud2 mrpt2ros");
                                                   ^
[ 50%] Building CXX object mrpt_navigation-0.1.8/mrpt_bridge/CMakeFiles/mrpt_bridge.dir/src/laser_scan.cpp.o
/home/lvchaoshun/bzrobot_ws/bz_external/mrpt_navigation-0.1.8/mrpt_bridge/src/laser_scan.cpp: In function 'bool mrpt_bridge::convert(const LaserScan&, const mrpt::poses::CPose3D&, mrpt::obs::CObservation2DRangeScan&)':
/home/lvchaoshun/bzrobot_ws/bz_external/mrpt_navigation-0.1.8/mrpt_bridge/src/laser_scan.cpp:35:12: error: 'struct mrpt::utils::ContainerReadOnlyProxyAccessor<std::vector<float> >' has no member named 'resize'
  _obj.scan.resize(N);
            ^
/home/lvchaoshun/bzrobot_ws/bz_external/mrpt_navigation-0.1.8/mrpt_bridge/src/laser_scan.cpp:36:18: error: 'struct mrpt::utils::ContainerReadOnlyProxyAccessor<std::vector<char> >' has no member named 'resize'
  _obj.validRange.resize(N);
                  ^
/home/lvchaoshun/bzrobot_ws/bz_external/mrpt_navigation-0.1.8/mrpt_bridge/src/laser_scan.cpp:45:21: error: assignment of read-only location '_obj.mrpt::obs::CObservation2DRangeScan::scan.mrpt::utils::ContainerReadOnlyProxyAccessor<STLCONTAINER>::operator[]<std::vector<float> >(((int)i_mrpt))'
   _obj.scan[i_mrpt] = _msg.ranges[i_ros];
                     ^
/home/lvchaoshun/bzrobot_ws/bz_external/mrpt_navigation-0.1.8/mrpt_bridge/src/laser_scan.cpp:46:27: error: assignment of read-only location '_obj.mrpt::obs::CObservation2DRangeScan::validRange.mrpt::utils::ContainerReadOnlyProxyAccessor<STLCONTAINER>::operator[]<std::vector<char> >(((int)i_mrpt))'
   _obj.validRange[i_mrpt] = ((_obj.scan[i_mrpt] < (_msg.range_max*0.95)) && (_obj.scan[i_mrpt] > _msg.range_min)) ? 1:0;
                           ^
mrpt_navigation-0.1.8/mrpt_bridge/CMakeFiles/mrpt_bridge.dir/build.make:134: recipe for target 'mrpt_navigation-0.1.8/mrpt_bridge/CMakeFiles/mrpt_bridge.dir/src/laser_scan.cpp.o' failed
make[2]: *** [mrpt_navigation-0.1.8/mrpt_bridge/CMakeFiles/mrpt_bridge.dir/src/laser_scan.cpp.o] Error 1
make[1]: *** [mrpt_navigation-0.1.8/mrpt_bridge/CMakeFiles/mrpt_bridge.dir/all] Error 2
make: *** [all] Error 2
CMakeFiles/Makefile2:3575: recipe for target 'mrpt_navigation-0.1.8/mrpt_bridge/CMakeFiles/mrpt_bridge.dir/all' failed
Makefile:138: recipe for target 'all' failed
20:15:18: 进程"/usr/bin/make"退出，退出代码 2 。
Error while building/deploying project bzrobot_external (kit: 桌面)
When executing step "Make"

jlblancoc, 2017-09-16:
I'm closing this one, and we can go on the discussion in #86 ok? 

Anyway: our current version policy is that ROS master should build correctly against MRPT 1.5.4 (which right now is the branch `mrpt-1.5` in the MRPT repo) **or** the `master` branch (right now 1.9.9, eventually, 2.0). 

Any error building against those two versions, indeed, is a bug and we should fix it.

closed 2017-09-16 by jlblancoc