mrpt-ros-pkg/mrpt_navigation #87: Version don't match report

lvchaoshun, 2017-09-16:
I install the MRPT master from source on debian 9 stretch.
$ mkdir build  
$ cd build 
$ cmake ..  
$ make j4
$ sudo make install

And then I clone the mrpt_navigation-master:
$ cd ~/catkin_ws/src
$ git clone https://github.com/mrpt-ros-pkg/mrpt_navigation
$ cd ..
$ catkin_make

**Output:**
-- ==> add_subdirectory(mrpt_navigation-master/mrpt_reactivenav2d)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- MRPT_VERSION: 1.9.9
-- MRPT_INCLUDE_DIRS: /usr/include/eigen3;/opt/ros/lunar/include;/usr/local/include/mrpt/mrpt-config/;/usr/include/suitesparse;/usr/local/include/mrpt/base/include;/usr/local/include/mrpt/graphs/include;/usr/local/include/mrpt/maps/include;/usr/local/include/mrpt/kinematics/include;/usr/local/include/mrpt/obs/include;/usr/local/include/mrpt/opengl/include;/usr/local/include/mrpt/nav/include
-- +++ processing catkin package: 'pose_cov_ops'
-- ==> add_subdirectory(pose_cov_ops-master)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
CMake Warning at /opt/ros/lunar/share/catkin/cmake/catkin_package.cmake:166 (message):
  catkin_package() DEPENDS on 'mrpt' but neither 'mrpt_INCLUDE_DIRS' nor
  'mrpt_LIBRARIES' is defined.
Call Stack (most recent call first):
  /opt/ros/lunar/share/catkin/cmake/catkin_package.cmake:102 (_catkin_package)
  pose_cov_ops-master/CMakeLists.txt:85 (catkin_package)

-- +++ processing catkin package: 'mrpt_localization'
-- ==> add_subdirectory(mrpt_navigation-master/mrpt_localization)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Configuring done
CMake Error at mrpt_navigation-master/mrpt_bridge/CMakeLists.txt:89 (add_dependencies):
  The dependency target "mrpt-base" of target "test_Bridge" does not exist.
CMake Error at mrpt_navigation-master/mrpt_bridge/CMakeLists.txt:89 (add_dependencies):
  The dependency target "mrpt-graphs" of target "test_Bridge" does not exist.
CMake Error at mrpt_navigation-master/mrpt_bridge/CMakeLists.txt:89 (add_dependencies):
  The dependency target "mrpt-maps" of target "test_Bridge" does not exist.
CMake Error at mrpt_navigation-master/mrpt_bridge/CMakeLists.txt:89 (add_dependencies):
  The dependency target "mrpt-obs" of target "test_Bridge" does not exist.
CMake Error at mrpt_navigation-master/mrpt_bridge/CMakeLists.txt:89 (add_dependencies):
  The dependency target "mrpt-opengl" of target "test_Bridge" does not exist.
CMake Error at mrpt_navigation-master/mrpt_bridge/CMakeLists.txt:89 (add_dependencies):
  The dependency target "mrpt-slam" of target "test_Bridge" does not exist.
CMake Error at mrpt_navigation-master/mrpt_bridge/CMakeLists.txt:89 (add_dependencies):
  The dependency target "mrpt-tfest" of target "test_Bridge" does not exist.
CMake Error at mrpt_navigation-master/mrpt_bridge/CMakeLists.txt:89 (add_dependencies):
  The dependency target "mrpt-vision" of target "test_Bridge" does not exist.

-- Generating done
-- Build files have been written to: /home/lvchaoshun/catkin_ws/build
Invoking "cmake" failed

I think the mrpt master version don't match the mrpt_navigation master version.

So, anyone can help me? 
Best regards!

jlblancoc, 2017-09-16:
Hi, 

We're really busy these weeks updating all the ROS packages to get them in synch with MRPT master. But in theory, MRPT master should match ROS master. 

I **strongly** advice against running `sudo make install`, not only for MRPT but in general. 
Please, go to your MRPT build directory and run `sudo make uninstall`; then go to `/usr/local` and make sure you don't have duplicated versions of MRPT (or other libs?). 

Then, run `dpkg --list | grep mrpt` to see if you also have any MRPT debian package installed. 

When all is clean, then just build MRPT in its build directory, and set the environment variable:

`export MRPT_DIR=/path/to/mrpt/build`

it could go into `.bashrc`, next to other ROS variables. In this way, you'll **make sure** CMake will always find the correct MRPT version.

If this still fails: what version of ROS are you using?

lvchaoshun, 2017-09-17:
MRPT is installed in my computer first time. 
My environment is Debian 9 stretch, ROS lunar.
And I used your method, the result is the same with before.

Is the MRPT don't support ROS lunar?

jlblancoc, 2017-09-17:
Thanks for the report!

That might be the problem: the latest version on which we tested mrpt ros
packages is ros kinetic.

Will investigate it. Any idea is welcome!

lvchaoshun, 2017-09-17:
Thank you very much.

closed 2017-09-19 by lvchaoshun