mrpt-ros-pkg/mrpt_navigation #98: mrpt_reactivenav2d does not generate velocity commands

MariotoA, 2018-04-25:
Hi,

I have been trying to launch [mrpt_reactivenav2d](https://github.com/mrpt-ros-pkg/mrpt_navigation/tree/master/mrpt_reactivenav2d) using [stage_ros](https://github.com/ros-gbp/stage-release/tree/release/kinetic/stage) + fake_navigation + map_server. (The reason being, mvsim does not seem to be compatible with MRPT 1.9.9 >= yet) 

So, every node starts ok, but when I send a goal to the navigator (from a point clicked in rviz) I get an error and navigation is cancelled due to a timeout.

These are the tag nodes I think are relevant. My guess is the problem lies in the ".ini" files in the package itself since stage responds normally to my keyboard commands and the vehicle shape is received by the reactive node without trouble. Note that I am using the config files it provides the repository. Or maybe mvsim plays a role that I am not accounting for.
```xml

<node name="map_server" pkg="map_server" type="map_server" args="$(arg map_file)" launch-prefix="$(arg launch_prefix)" />

<node pkg="fake_localization" type="fake_localization" name="fake_localization" output="screen">	    
	    <param name="odom_frame_id" value="odom"/>
	    <param name="global_frame_id" value="/map"/>
	    <param name="base_frame_id" value="base_link"/>
</node>

<node pkg="stage_ros" type="stageros" name="stage_ros" respawn="false" output="screen" args="$(arg world_file)" >
    <param name="laser_scan_topic" value="/laser_scan"/>
    <param name="laser_frame_id" value="/laser_stage" />
    <param name="disable_odometry" value="$(arg disable_odometry)" />  #if true, no topic, neither tf transform are published for odometry
    <param name="odom_topic" value="/odom"/>
    <param name="odom_frame_id" value="/odom" />
</node> 

<node pkg="mrpt_local_obstacles" type="mrpt_local_obstacles_node" name="mrpt_local_obstacles_node" output="screen">
		<param name="source_topics_2dscan" value="laser_scan"/>
		<param name="show_gui" value="false"/>
		<param name="time_window" value="60"/>
</node>

<!-- 
I'm publishing the robot shape in a very simple node found here:
https://github.com/MariotoA/shape_publisher
-->
<node pkg="shape_publisher" type="shape_publisher" name="shape_publisher" output="screen" launch-prefix="xterm -e"/>

<node pkg="mrpt_reactivenav2d" type="mrpt_reactivenav2d_node" name="mrpt_reactivenav2d_node" output="screen">
<!-- *Important*: This external config file holds the most important navigation settings --> 
	<param name="cfg_file_reactive" value="$(find mrpt_reactivenav2d)/tutorial/reactive2d_config.ini"/> 
	<!-- This is to allow the reactive responsive to RVIZ GUI commands -->
	<remap from="reactive_nav_goal" to="/move_base_simple/goal" />
	<!-- Enable logs to ~/.ros/reactivenav.logs , use navlog-viewer to open them -->
	<!--<param name="save_nav_log" value="true"/> -->
	<param name="topic_robot_shape" value="/chassis_polygon" /> 
</node>

<node name="rviz_player" pkg="rviz" type="rviz" args="-d $(find tfg_reactive_pkg)/rviz/simbot_reactive_only.rviz"/>

```

This is the error trace itself:

[ INFO] [1524665257.040733976, 7.600000000]: Nav target received via topic sub: (2.555,1.869, -23.832deg) [frame_id=/map]
[ INFO] [1524665257.041667984, 7.600000000]: [navigateTo] Starting navigation to [2.554826 1.868601 -23.831614]
[MRPT_navigator|INFO |16:07:37.0436] [CAbstractNavigator::navigationStep()] Starting Navigation. Watchdog initiated...
**[MRPT_navigator|DEBUG|16:07:37.0441] [CAbstractNavigator::navigationStep()] Navigation Params:
navparams. Single target:
target_coords = (0.000,0.000,0.000 deg)
target_frame_id = "map"
targetAllowedDistance = 0.500
targetIsRelative = NO
targetIsIntermediaryWaypoint = NO
targetDesiredRelSpeed = 0.05
multiple_targets:
target[0]:
target_coords = (2.555,1.869,0.000 deg)
target_frame_id = "map"
targetAllowedDistance = 0.400
targetIsRelative = NO
targetIsIntermediaryWaypoint = NO
targetDesiredRelSpeed = 0.05
restrict_PTG_indices: []**

[MRPT_navigator|DEBUG|16:07:37.0530] CWaypointsNavigator::checkHasReachedTarget() called with targetDist=3.60555 return=0 **waypoint: wps.timestamp_nav_started=INVALID_TIMESTAMP**
[MRPT_navigator|INFO |16:07:37.0566] [CReactiveNavigationSystem::STEP1_InitPTGs] Initializing PTG#0 (`CPTG_DiffDrive_C,K=1`)...

[CPTG_DiffDrive_CollisionGridBased::initialize] Starting... *** THIS MAY TAKE A WHILE, BUT MUST BE COMPUTED ONLY ONCE!! **
Initializing PTG './ReacNavGrid_000.dat.gz'...loaded from file OK
[MRPT_navigator|INFO |16:07:41.0283] Done!
[MRPT_navigator|INFO |16:07:41.0287] [CReactiveNavigationSystem::STEP1_InitPTGs] Initializing PTG#1 (`CPTG_DiffDrive_alpha,av=57deg,aw=57deg`)...

[CPTG_DiffDrive_CollisionGridBased::initialize] Starting... *** THIS MAY TAKE A WHILE, BUT MUST BE COMPUTED ONLY ONCE!! **
Initializing PTG './ReacNavGrid_001.dat.gz'...loaded from file OK
[MRPT_navigator|INFO |16:07:50.2053] Done!
[MRPT_navigator|INFO |16:07:50.2057] [CReactiveNavigationSystem::STEP1_InitPTGs] Initializing PTG#2 (`CPTG_DiffDrive_C,K=-1`)...

[CPTG_DiffDrive_CollisionGridBased::initialize] Starting... *** THIS MAY TAKE A WHILE, BUT MUST BE COMPUTED ONLY ONCE!! **
Initializing PTG './ReacNavGrid_002.dat.gz'...loaded from file OK
[MRPT_navigator|INFO |16:07:54.4521] Done!
[MRPT_navigator|ERROR|16:08:32.6288] [CAbstractPTGBasedReactive::performNavigationStep] Stopping robot and finishing navigation due to exception:

 =============== MRPT EXCEPTION =============
void mrpt::expr::CRuntimeCompiledExpression::compile(const string&, const std::map<std::__cxx11::basic_string<char>, double>&, const string&), line 71:
Error compiling expression (name=`score: collision_free_distance`): `var effective_trg_d_norm := max(0,target_d_norm-move_cur_d); 	if (collision_free_distance>(effective_trg_d_norm+0.05), 		1.0, 		collision_free_distance)`. Error: `ERR186 - Undefined symbol: 'target_d_norm'`:
[MRPT_navigator|DEBUG|16:08:32.8244] CWaypointsNavigator::checkHasReachedTarget() called with targetDist=3.60555 return=0 waypoint: wps.timestamp_nav_started=INVALID_TIMESTAMP
[MRPT_navigator|WARN |16:08:32.8289] Timeout approaching the target. Aborting navigation.
[MRPT_navigator|ERROR|16:08:33.0357] [CAbstractNavigator::navigationStep()] Stopping Navigation due to a NAV_ERROR state!

--------------------------------------------------------------------------------------------------------

I am currently using xubuntu-16.04, MRPT 1.9.9 (I think. I followed [this](https://www.mrpt.org/MRPT_in_GNU/Linux_repositories) and mrpt/utils does not exist for example) and ros-kinetic. 

Also, I noticed just now this is the same bug as [this one](https://www.bountysource.com/issues/49772103-mrpt-exception-at-mrpt-math-cruntimecompiledexpression-compile), but I get two navparams for some reason, one being, I assume, the robot pose and the other the goal I send.

I've emphasized in bold text the parts of the trace that seems suspicious to me. I did not include any map files since I believe the problem does not come from there, but maybe i'm mistaken, just ask if you need them.

Any idea? By the way, newbie here, so insights or explanations are well appreciated. Thanks a lot.

Best regards.

jlblancoc, 2018-04-27:
Hi, 
I don't have time to fix this now, but I'm almost 100% sure the problem is in the .ini file for the reactive navigation. Please, compare the content in this file with the one found in the main MRPT repository, under  `share/mrpt/config_files/` reactivenav / ptgs.

MariotoA, 2018-08-08:
Hi @jlblancoc,

Yes, I was able to work around the bug. I'm not sure if what happened to me had something to do with the processing of targets though.

What happened to me was, since trajectories are generated at maximum linear velocity, I was sending targets too close to the base. It seemed all trajectories were discarded and I got a compiled expression exception, for some reason the exception only occurs if navigation hadn't start before, after that the DEBUG message yields PTG "-1" is selected. Lowering the maximum linear vels in the PTGs in the .ini files solved this issue.

However, I don't think this relates to you since your target is at, I assume:

> target_coords = (11.051,0.585,0.000 deg)

It also, at least for me, seems like navigation never truly ends and the navigator forever sends null velocity commands when the goal is reached. This only happens when [`navigate`](https://github.com/MRPT/mrpt/blob/master/libs/nav/src/reactive/CAbstractNavigator.cpp#L314) method is used, I tried a version with [`navigateWaypoints`](https://github.com/MRPT/mrpt/blob/master/libs/nav/src/reactive/CWaypointsNavigator.cpp#L67) and navigation ends fine for me.

It is still a mistery why the origin is sent as a target.

Sorry for the monthly delay.

Regards.

JackRen88, 2021-07-20:
Hello, @MariotoA .
Which version of mrpt_navigation and mrpt are you using？

jlblancoc, 2022-07-01:
For the latest releases of all packages (mvsim + mrpt_*), the roslaunch examples work straight away, so we will close this one.
Cheers