mrpt-ros-pkg/mrpt_navigation #100: Cannot release with bloom to ROS melodic

jlblancoc, 2018-04-27:
Apparently, melodic is still missing some other deps: 

```
Pre-verifying Debian dependency keys...
Running 'rosdep update'...
Could not resolve rosdep key 'cv_bridge'
Failed to resolve cv_bridge on ubuntu:artful with: Error running generator: Failed to resolve rosdep key 'cv_bridge', aborting.
cv_bridge is depended on by these packages: ['mrpt_bridge']
<== Failed
Could not resolve rosdep key 'cv_bridge'
Failed to resolve cv_bridge on ubuntu:bionic with: Error running generator: Failed to resolve rosdep key 'cv_bridge', aborting.
cv_bridge is depended on by these packages: ['mrpt_bridge']
<== Failed
Could not resolve rosdep key 'marker_msgs'
Failed to resolve marker_msgs on ubuntu:artful with: Error running generator: Failed to resolve rosdep key 'marker_msgs', aborting.
marker_msgs is depended on by these packages: ['mrpt_bridge', 'mrpt_rawlog']
<== Failed
Could not resolve rosdep key 'marker_msgs'
Failed to resolve marker_msgs on ubuntu:bionic with: Error running generator: Failed to resolve rosdep key 'marker_msgs', aborting.
marker_msgs is depended on by these packages: ['mrpt_bridge', 'mrpt_rawlog']
<== Failed
Could not resolve rosdep key 'pcl_conversions'
Failed to resolve pcl_conversions on ubuntu:artful with: Error running generator: Failed to resolve rosdep key 'pcl_conversions', aborting.
pcl_conversions is depended on by these packages: ['mrpt_bridge']
<== Failed
Could not resolve rosdep key 'pcl_conversions'
Failed to resolve pcl_conversions on ubuntu:bionic with: Error running generator: Failed to resolve rosdep key 'pcl_conversions', aborting.
pcl_conversions is depended on by these packages: ['mrpt_bridge']
<== Failed
Could not resolve rosdep key 'pose_cov_ops'
Failed to resolve pose_cov_ops on ubuntu:artful with: Error running generator: Failed to resolve rosdep key 'pose_cov_ops', aborting.
pose_cov_ops is depended on by these packages: ['mrpt_localization']
<== Failed
Could not resolve rosdep key 'pose_cov_ops'
Failed to resolve pose_cov_ops on ubuntu:bionic with: Error running generator: Failed to resolve rosdep key 'pose_cov_ops', aborting.
pose_cov_ops is depended on by these packages: ['mrpt_localization']
<== Failed
Could not resolve rosdep key 'teleop_twist_keyboard'
Failed to resolve teleop_twist_keyboard on ubuntu:artful with: Error running generator: Failed to resolve rosdep key 'teleop_twist_keyboard', aborting.
teleop_twist_keyboard is depended on by these packages: ['mrpt_tutorials']
<== Failed
Could not resolve rosdep key 'teleop_twist_keyboard'
Failed to resolve teleop_twist_keyboard on ubuntu:bionic with: Error running generator: Failed to resolve rosdep key 'teleop_twist_keyboard', aborting.
teleop_twist_keyboard is depended on by these packages: ['mrpt_tutorials']
```

pose_cov_ops is ours. For the rest, I guess we should wait.

jolting, 2018-04-30:
I think this is an example of how to release mrpt for ros. 
https://github.com/intel-ros/librealsense-release

jlblancoc, 2018-04-30:
Hmm... I think I don't get it... MRPT is already released using bloom like those pkgs:
https://github.com/mrpt-ros-pkg-release

The problem is that released packages must be built against deb pkgs in the official Ubuntu releases... and mrpt is too old in those ones! :-(

jolting, 2018-04-30:
For example opencv3 wasn't shipped in xenial, but ROS used it.

You need to package up libmrpt as a ROS package.

It's the only alternative to using the system packaged version.

jlblancoc, 2018-05-01:
FYI: related PR is now proposed here: https://github.com/ros/rosdistro/pull/17692/files

jlblancoc, 2018-05-04:
mrtp1 is now into ROS kinetic, but it has problems building .deb packages: 
Search for `error:` in http://build.ros.org/job/Kbin_uX64__mrpt1__ubuntu_xenial_amd64__binary/3/console

Seems a consecuence of stripping rpath, but must look at it more carefully... MRPT's own debian/* scripts are successful in building .deb packages, so it must be something in the way ROS .deb pkgs are built (?). 
My first reaction is trying to add "rpath" via CMake... (?)

Any other idea?

PS:  Just compiling works fine: http://build.ros.org/job/Kdev__mrpt1__ubuntu_xenial_amd64/  It's building .deb pkgs what is wrong...

PS2: Same issue from another guy a few days ago here: https://answers.ros.org/question/290073/problem-building-with-non-installed-dependencies/#

jlblancoc, 2018-05-04:
Comparing to opencv3 (successful) build logs, I cannot find a difference in neither where LIBS are getting installed, nor in the dpkg-shlibdeps command... There must be some subtle difference...

jolting, 2018-05-05:
Did you notice that there are two versions in that log?
1.5.6 and 1.5.7

jlblancoc, 2018-05-05:
Interesting.... :-)
Will try to fix it in a new release and see if that makes the ROS farm
happy.

jlblancoc, 2018-05-06:
After more digging, looks like we missed to follow all the REP-136 specifications.
http://www.ros.org/reps/rep-0136.html
Will try to fix it...

jlblancoc, 2018-05-13:
Last chapter of this novel: 
https://github.com/ros-infrastructure/ros_buildfarm/issues/543

jolting, 2018-05-14:
Might need to add rpath.

https://gitlab.kitware.com/cmake/community/wikis/doc/cmake/RPATH-handling

jlblancoc, 2018-05-14:
> Might need to add rpath.

Sure, rpath could fix it... but per Debian policies it cannot be used, though... https://lintian.debian.org/tags/binary-or-shlib-defines-rpath.html

In fact, rpath's are strip during normal .deb generation. 

Also, the same sources (but with different versions of cmake, deb-helper pkgs, etc.) succeed in building .deb either manually (on my Xenial machine) or in the ROS buildfarms for Bionic. So, there must be something else... 

This is really frustrating!

jlblancoc, 2018-06-08:
Fixed, at last! 
Not 100% proud of the solution, but it just works: https://github.com/mrpt-ros-pkg-release/mrpt1-release/commit/17361d75ef9e73c3496dca74813c4b9ee90ec643

closed 2018-06-08 by jlblancoc