mrpt-ros-pkg/mrpt_navigation #101: refactor moving mrtp_bridge is breaking kinetic builds

tfoote, 2018-06-06:
I think that this change is causing https://github.com/ros/rosdistro/pull/17692

Is causing most of the mrpt packages on this list to fail to build: http://repositories.ros.org/status_page/ros_kinetic_default.html?q=REGRESSION

With tracebacks like this:
```
23:15:51 Looking for the '.dsc' file of package 'ros-kinetic-mvsim' with version '0.1.2-0'
23:15:52 Traceback (most recent call last):
23:15:52   File "/usr/lib/python3/dist-packages/apt/cache.py", line 194, in __getitem__
23:15:52     return self._weakref[key]
23:15:52   File "/usr/lib/python3.5/weakref.py", line 131, in __getitem__
23:15:52     o = self.data[key]()
23:15:52 KeyError: 'ros-kinetic-mrpt-bridge'
23:15:52 
23:15:52 During handling of the above exception, another exception occurred:
23:15:52 
23:15:52 Traceback (most recent call last):
23:15:52   File "/usr/lib/python3/dist-packages/apt/cache.py", line 198, in __getitem__
23:15:52     rawpkg = self._cache[key]
23:15:52 KeyError: 'ros-kinetic-mrpt-bridge'
```

jlblancoc, 2018-06-06:
Thanks for the ping!

Actually, any progress in Kinetic was blocked a few weeks ago due to this hard-to-fix bug found during the release of the new package `mrpt1` on which all these packages depend on... (sigh)

https://github.com/ros-infrastructure/ros_buildfarm/issues/543

Quite frustratingly, exactly the same release builds like a charm in Melodic.

If the regression has implications to Kinetic releases as a whole (they delay the launch of updates,...), let me know and I would even consider rolling back and freezing all `mrpt-*` packages to an older version for Kinetic, then move on with Melodic...

jlblancoc, 2018-06-08:
Finally fixed, Yay! 

http://build.ros.org/job/Kbin_uX64__mrpt1__ubuntu_xenial_amd64__binary/42/
http://repositories.ros.org/status_page/ros_kinetic_default.html?q=REGRESSION

closed 2018-06-08 by jlblancoc