mrpt-ros-pkg/mrpt_navigation #102: Segfault in mrpt_localization_node

fk011, 2018-07-12:
Hi,

I am using mrpt 1.9 (the latest version from github) in conjunction with mrpt_navigation. Upon running the launch file demo.launch in mrpt_localization the application segfaults. The segfault happens in mrpt_localization.cpp at line 316 where the dynamic cast is performed. More specifically, this happens during the eigen handmade_aligned_free call in Eigens Memory.h file. I am guessing something is wrongly allocated. 
Commenting out the show3DDebug call and omitting this issue allows the application to finish without any further crashes.
Could you please resolve this issue?
Many Thanks.

fk011, 2018-07-12:
I am adding another issue here and also correct my statement made above.
The application still crashes with an eigen unaligned array assert exception even if the line 316 in mrpt_localization.cpp is commented out (for testing purposes).

This error is not reproducible everytime the application is started, it can happen that the application finishes without any crash, so you might need to start it a few times. I guess this is because it can happen that the alignment is correct per chance.

I do not know how to properly fix the code so that it doesn't crash. The application can crash at any point where an Eigen class is used, if it crashes at one point in the code a next crash at the same piece of code is not guaranteed.
Any help is appreciated.

Best regards.

jlblancoc, 2018-07-12:
What's your OS? Ubuntu? 32, 64bit, arm?

It looks like a missing macro to enforce alignment by overloading the "new"
operator, but in theory that should affect 32bit only...

fk011, 2018-07-16:
Dear Professor,

so far, this error has been reproduced on 3 separate machines, one with an i7-4790, one with an i7-4700 and a laptop with an i5-8350U CPU.

The issue seems to be the manual compilation of mrpt2.0, whereby the code is optimized differently than in mrpt-ros-pkg.

Adding the compiler flag for the avx instruction set seems to solve the problem on those machines.
I added add_compile_options(-march=native) at the beginning of the CMakeLists.txt of mrpt_localization and the program does not crash anymore.

jlblancoc, 2018-07-16:
Hmmm... interesting.  Thanks for sorting it out 👍 

Perhaps the best solution is to make MRPT to export that flag via MRPTConfig.cmake, so all is consistent regarding whether the flag is used or not all across the different projects. 

Added: https://github.com/MRPT/mrpt/issues/800

(cc: @jolting )

closed 2018-07-16 by jlblancoc

jolting, 2018-07-16:
I agree with exporting the required flags in MRPTConfig.cmake

We should avoid using the native optimization flag though because this can have disastrous results. This has a different behavior depending on the build machine and the machine you're running on. AVX-512 is now supported on some newer hardware and we don't want that accidentally turning on. AVX instructions aren't supported by some older 64-bit machines.

AVX instructions mean that the minimum requirement is Gen2 Intel and beyond.. I believe SSE is safe for all machines.