mrpt-ros-pkg/mrpt_navigation #107: How to use mrpt_localization with limited memory?

Melmarkt, 2019-01-09:
Hi!
I'm trying to implement mrpt-localization (PF mode), on a single board computer. Actually, I  can load a .pgm/.yaml map and localize successfully my robot on it with mrpt-localization, but after 10 minutes my RAM is saturated, and my system freeze.

[image]

Do you have any advice or solution for this problem?

Thanks!

jlblancoc, 2019-01-10:
Hi!

Do you mean... the memory usage is continuously increasing? Please confirm if that's the case, since it would mean we have a memory leak, either on the ROS side or in MRPT itself!

Please, let us know as well what version of MRPT are you using.

Melmarkt, 2019-01-10:
Yes exactly, the memory usage is continuously increasing. 
I use the version of MRPT which is installed with ros-kinetic-mrpt-navigation (0.1.23) from the apt-get server, and I installed only that.

 My system is under Ubuntu 16.04.5 LTS (GNU/Linux 4.9.80-v7+ armv7l), and I use ROS Kinetic (1.12.14)

jlblancoc, 2019-03-04:
Hi @Melmarkt , 

I tried to reproduce this bug today (with the mrpt-1.5 branch built locally on an Ubuntu 18.04), but the memory use was not increasing... 
Could you try running your localization node with valgrind to see if it shows something interesting? 

http://wiki.ros.org/roslaunch/Tutorials/Roslaunch%20Nodes%20in%20Valgrind%20or%20GDB

If you do that, disable the "mrpt-gui" of pf-localization, since wxWidgets (and the OpenGL subsystem) is known to generate **lots** of valgrind warnings and that would make much harder to find a real leak...

Melmarkt, 2019-03-28:
> ping...

Hi,
Sorry, i changed of jobs since 2 weeks, so I need to rebuild my setup
first. But i will try as soon as it's done. (That could take some times, I
still need to receive some parts)
Sorry for the late answer.

jlblancoc, 2019-03-28:
No problem at all, thanks!

closed 2020-09-23 by jlblancoc