mrpt-ros-pkg/mrpt_navigation #110: Missing mvsim dependency found in .launch

andersfischernielsen, 2019-04-02:
I'm currently doing some research on static analysis of ROS packages at the IT University of Copenhagen, more specifically dependency checking of `.launch` files in ROS packages. 

There seems to be a `$(find mvsim)` in  
- `mrpt_navigation/mrpt_local_obstacles/launch/demo_with_mvsim.launch` 
    - that's not declared in `mrpt_local_obstacles/package.xml`
- `mrpt_navigation/mrpt_reactivenav2d/launch/reactive_nav_demo_with_mvsim.launch` 
    - that's not declared in `mrpt_reactivenav2d/package.xml`

Installing `mrpt_navigation` without having `mvsim` installed would make `roslaunch` fail on these two `.launch` files, right? 

I'm unsure whether this is a dependency bug, since there are only two specific launch configurations using `mvsim` and not the entire package(s). 
Nonetheless, `roslaunch` would fail, no?

jlblancoc, 2019-04-04:
Hi @andersfischernielsen, 

Thanks for the valuable feedback! 
Well... yes: I guess that mvsim could, ad *should*, be listed as a dep. in the package.xml. 
OTOH, it's "only" used for demo purposes: this package can work perfectly without it, hence that it was not listed as a build/runtime dependency. 

If we had something like "Recommends:" for Debian dependencies, I guess mvsim would perfectly fit in that category  :-)

I'm not sure about what to do here... probably leaving it unlisted would be the "best" in terms of maintenance problems, i.e. if someday mvsim "breaks" (FTBFS, etc.) it would not prevent this one to keep being a valid ROS package... do you agree?

Cheers.

andersfischernielsen, 2019-04-05:
It's hard to tell in my opinion, since (like you say) it is just for demo purposes. 

It would therefore be slightly annoying for users that never run the demo to have an extra unneeded dependency installed on their system and what you're saying about having a dependency (`mvsim`) that could break the package down the would also be a problem. 

On the other hand, if you really **do** want to run the demo (because you'd want to use `mvsim`), then it would be annoying to manually have to install the missing dependency. 

It's up to you really. I don't know a lot about this package specifically, I was just running the `.launch` dependency checker I've developed on this package and saw the results I described in the first comment. :)

jlblancoc, 2019-04-05:
Ok! Thanks for your work... any attempt to automate package checking in ROS is welcome, so thanks for your effort :+1:

Just like in https://github.com/mrpt-ros-pkg/mrpt_slam/issues/57, I think we'll just leave that dep. out, for now... mainly because I don't trust I will ever have time to maintain so many packages, and mvsim has a low priority ;-) 

Cheers

closed 2019-04-05 by jlblancoc