mrpt-ros-pkg/mrpt_navigation #112: Cannot convert bag to rawlog

aaravrav142, 2019-08-22:
Hi, 

I am using the mrpt 1.5 installed from PPA on a fresh ubuntu 16.04 install with ROS  kinetic. 
I have all the mrpt ros packages in my catkin workspace and have them build successfully. 
The MRPT is working as I can open the sample apps like RawLogViewer etc from the command line. 

I am following the the tutorial to convert the given bag filte to the rawlog format for further analysis. The bag file comes with the mrpt ros packages inside the mrpt_rawlog/tutorial folder (driving_in_office.bag). However, the rawlog does not get saved properly. 
I always get stream of continuous info messages as 
`
$rosrun mrpt_rawlog rawlog_record_node scan:=/r1/front_laser/scan _odom_frame_id:=r1/odom _base_frame_id:=r1/base_link _raw_log_folder:=/home/arav/Documents/`

```
 Problem to syn data for sensor frame!
 Problem to syn data for sensor frame!
 Problem to syn data for sensor frame!
writing dataset to disk...
INFO: pRawLog    entries 0
INFO: pRawLogASF entries 0
```
I have tried this with other bag files containing topics such as scan, odom and tf but getting same results as below:
`$ rosrun mrpt_rawlog rawlog_record_node scan:=/scan _odom_frame_id:=odom _base_frame_id:=/base_link _raw_log_folder:=/home/arav/Documents/`
 ```
INFO: pRawLog    entries 1093
INFO: pRawLogASF entries 260
INFO: write _home_arav_Documents_/2019_08_22,23_35_40.384428ros_log.rawlog
ERROR: Error writing to _home_arav_Documents_/2019_08_22,23_35_40.384428ros_log.rawlog
INFO: write _home_arav_Documents_/2019_08_22,23_35_40.384428ros_log_asf.rawlog
ERROR: Error writing to _home_arav_Documents_/2019_08_22,23_35_40.384428ros_log_asf.rawlog

```
I have successfully used it before when I was using ROS indigo with the ubuntu 14 but on Kinetic this does not work. All our PCs were upgraded to ubuntu 16 so I cannot revert back now. Tested it with MRPT 1.9 as well and getting same result. 

Can you help with this because I urgently need to analyse the rawlog to finish my project. 

Thanks

hoonkai, 2019-09-16:
I get  a similar problem too. At first, it gives the following error:

```
[ INFO] [1568627085.505680852]: record_bearing_range: false
[ INFO] [1568627085.506049965]: record_beacon_range: false
[ INFO] [1568627085.509747725]: rawlog file: 2019_09_16,19_44_45.509685ros_log.rawlog
[ INFO] [1568627105.451329755]: getStaticTF
[ERROR] [1568627105.451359238]: "r1/base_link" passed to lookupTransform argument target_frame does not exist. 
```

hoonkai, 2019-09-16:
I should note that I get that error using the latest branch, melodic, MRPT 1.5, Ubuntu 18.04. I get exactly like @aaravrav142 on 0.1.24 though.

hoonkai, 2019-09-16:
Fixed. I just had to turn down the playback rate.

aaravrav142, 2019-09-18:
@hoonkai  did you get the rawlog generated in the end?

aaravrav142, 2020-01-29:
I am still facing the same problem with fresh install of mrpt 1.9 and mrpt ros pkgs. No msgs been recorded after playing the rosbag. Should I try melodic for a change?? Any help is deeply appreciated. 
Thanks

jlblancoc, 2022-07-01:
I will probably remove rawlog_record from this package in a subsequent release, since there is a more modern and flexible way to doing it via the upstream's app `rosbag2rawlog` from package Ubuntu native package `mrpt-apps` or from ROS package `ros-${ROS_DISTRO}-mrpt2`. 
To do: 
- [ ] Add proper docs and tutorial to https://docs.mrpt.org on `rosbag2rawlog`.
- [ ] Remove `rawlog_record ` from this repo.

jlblancoc, 2024-04-08:
For the records: 
- ROS1 -> .rawlog: use mrpt2/rosbag2rawlog 
- ROS2 -> .rawlog: use (mrpt_navigation/mrpt_rawlog](https://github.com/mrpt-ros-pkg/mrpt_navigation/tree/ros2/mrpt_rawlog) rosbag2rawlog version from

closed 2024-04-08 by jlblancoc