mrpt-ros-pkg/mrpt_navigation #113: C++17 syntax won't compile

hoonkai, 2019-09-16:
Hi

Looking at commit c9dcb7193843b89ef311806bbd2a185f2200dafa C++17 syntax is used, but MRPT<1.9.9 enables C++14 as per https://github.com/mrpt-ros-pkg/mrpt_navigation/blob/master/mrpt_rawlog/CMakeLists.txt#L12 Not sure if I'm missing anything.

Thanks

jlblancoc, 2019-09-16:
In theory, these ros packages built ok against mrpt >=1.9, but I'll check again. 
You are right in the inconsistency between c++14 and c++17, it should be fixed. 
However, mrpt>=1.9 uses exported cmake targets, which carries the requirement of using c++17 automatically just by declaring the need to link against them, so the line you pointed to could be actually entirely removed, to avoid the contradictory statements of using c++14 & c++17 simultaneously. 

You can run `VERBOSE=1 make -j1` to see the full build cmd line, and probably both flags are there, which is clearly undesirable, since it's not clear which one prevails...

jlblancoc, 2019-09-16:
Should be fixed now!

commit ac020ee2 (2019-09-16): fix build against mrpt1 (Closes #113)
closed 2019-09-16 by jlblancoc