mrpt-ros-pkg/mrpt_navigation #120: NaN in Odometry transform  leads to permanent TF NaN errors

seanrivera, 2020-03-20:
The odometry calculation of computeFromOdometry fails to check if the pose.pose.position or pose.pose.orientation is Nan before passing them into the core MRPT library. At which point the odometry is updated to an incorrect Nan and it propagates through the calculations.

jlblancoc, 2020-03-20:
Hi Sean, 
Is there any valid situation in which NaN is allowed to be published by the odometry? Shouldn't that be the first thing to fix instead?

seanrivera, 2020-03-23:
I do not believe there is any valid situation in which NaN can be published, however, there are many situations where another malfunctioning node can publish incorrect positions. I am presently performing a sensitivity analysis of some of the most commonly used nodes and am looking into the cascading effects of failures.