mrpt-ros-pkg/mrpt_navigation #125: need some Tuning methodology of pf-localization.ini file.

ZOUJIASHUAI, 2021-02-24:
can you tell me the meaning of these parameters in pf-localization.ini?

Gunz-katomaran, 2023-04-19:
Need help to understand and tune ,ini file for better localization @jlblancoc

jlblancoc, 2023-05-02:
I know, I know... documentation is scarce and spread here and there.

Some hints are given here: https://github.com/mrpt-ros-pkg/mrpt_navigation/issues/133#issuecomment-1531296754

You can also read about the different motion model parametes [in these docs](https://docs.mrpt.org/reference/latest/tutorial-motion-models.html).

But there are actually much more insights that need to be understood about the "adaptive" part of the number of particles (KLD paper, or the "A" in the famous "ACML"). 
Those parameters are in those in [mrpt::slam::TKLDParams](https://docs.mrpt.org/reference/latest/class_mrpt_slam_TKLDParams.html) and are fairly important to tune.

I'll try to put everything together in some nice piece of tutorial asap.