mrpt-ros-pkg/mrpt_navigation #128: dpkg error: trying to overwrite '/opt/ros/melodic/bin/RawLogViewer', which is also in package ros-melodic-mrpt1 1.5.9-1bionic.20210509.072520 dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)

SRai22, 2022-04-27:
When trying to perform rosdep install -y --from-paths . --ignore-src --rosdistro=melodic, tries to install both ros-melodic-mrpt1 and ros-melodic-mrpt2. Conflict arises for RawLogViewer and results in broken packages. 

Appreciate any help in this regard

SRai22, 2022-04-27:
I don't have a mrptq installed in the system. Also I changed all the CMakeLists to find mrpt 2. The error persists.

jlblancoc, 2022-04-28:
I found this to be caused by `mrpt_bridge`. I already submitted an updated version to rosdistro for Melodic. In the meanwhile that pkg can be built from sources. 

Also, in this (mrpt_navigation) repo, there's code still to be ported to the latest mrpt2. 
Try installing the pkg ros-melodic-mrpt2, then building from sources. 

I just fixed and committed a few corrections, but more are pending. If you see easily how to fix more, feel free of opening PRs if you do it earlier than myself! ;-)

jlblancoc, 2022-07-01:
This one was already fixed, thanks for reporting!

closed 2022-07-01 by jlblancoc