mrpt-ros-pkg/mrpt_navigation #129: Last release fails to build on ROS Noetic

sloretz, 2022-06-29:
It looks like the 1.0.0-1 release is failing to build on Noetic. The `Ndev` job is also failing. The next ROS Noetic sync will cause the binaries for these packages to no longer be available unless a new release of this repo is made with a fix.

It looks like the binary jobs for these packages are failing to build

* mrpt_local_obstacles
* mrpt_localization
* mrpt_map
* mrpt_rawlog
* mrpt_reactivenav2d

And the dev job is failing with this error

```
01:25:41 [ 50%] Building CXX object CMakeFiles/mrpt_local_obstacles_node.dir/src/mrpt_local_obstacles_node.cpp.o
01:25:41 [100%] Linking CXX executable /tmp/ws/devel_isolated/mrpt_local_obstacles/lib/mrpt_local_obstacles/mrpt_local_obstacles_node
01:25:53 /usr/bin/ld: CMakeFiles/mrpt_local_obstacles_node.dir/src/mrpt_local_obstacles_node.cpp.o: in function `LocalObstaclesNode::onNewSensor_Laser2D(boost::shared_ptr<sensor_msgs::LaserScan_<std::allocator<void> > const> const&)':
01:25:54 mrpt_local_obstacles_node.cpp:(.text._ZN18LocalObstaclesNode19onNewSensor_Laser2DERKN5boost10shared_ptrIKN11sensor_msgs10LaserScan_ISaIvEEEEE[_ZN18LocalObstaclesNode19onNewSensor_Laser2DERKN5boost10shared_ptrIKN11sensor_msgs10LaserScan_ISaIvEEEEE]+0x538): undefined reference to `mrpt::ros1bridge::fromROS(sensor_msgs::LaserScan_<std::allocator<void> > const&, mrpt::poses::CPose3D const&, mrpt::obs::CObservation2DRangeScan&)'
01:25:54 collect2: error: ld returned 1 exit status
```

http://repositories.ros.org/status_page/ros_noetic_default.html?q=REGRESSION
https://build.ros.org/job/Ndev__mrpt_navigation__ubuntu_focal_amd64/
https://build.ros.org/job/Nbin_uF64__mrpt_navigation__ubuntu_focal_amd64__binary/

jlblancoc, 2022-06-29:
Yeap, it's a bug in the mrpt2 package, I was waiting for the next mrpt2 version to be released, but the current one (already in melodic) would fix it. 

I wasn't sure about the tradeoff between being such a pain opening so many PRs to rosdistro vs. what problems the Jenkins failures might cause... From now on I'll de-prioritize my worries about being a pain ;-) 

Thanks!

jlblancoc, 2022-06-29:
First one here: https://github.com/ros/rosdistro/pull/33739

jlblancoc, 2022-07-01:
http://repositories.ros.org/status_page/ros_noetic_default.html?q=REGRESSION

Fixed now!

closed 2022-07-01 by jlblancoc