mrpt-ros-pkg/mrpt_navigation #131: mrpt localization node killed

Gunz-katomaran, 2022-10-25:
mrpt localization node killed on launch with following errors
`Oct 25 08:07:35 dotkat007 bash[25489]: PF_algorithm                            = pfStandardProposal
Oct 25 08:07:35 dotkat007 bash[25489]: m_resamplingMethod                      = prMultinomial
Oct 25 08:07:35 dotkat007 bash[25489]: adaptiveSampleSize                      = Y
Oct 25 08:07:35 dotkat007 bash[25489]: sampleSize                              = 1
Oct 25 08:07:35 dotkat007 bash[25489]: BETA                                    = 0.500000
Oct 25 08:07:35 dotkat007 bash[25489]: pfAuxFilterOptimal_MaximumSearchSamples = 10
Oct 25 08:07:35 dotkat007 bash[25489]: powFactor                               = 1.000000
Oct 25 08:07:35 dotkat007 bash[25489]: max_loglikelihood_dyn_range             = 15.000000
Oct 25 08:07:35 dotkat007 bash[25489]: pfAuxFilterStandard_FirstStageWeightsMonteCarlo = N
Oct 25 08:07:35 dotkat007 bash[25489]: pfAuxFilterOptimal_MLE                  = N
Oct 25 08:07:35 dotkat007 bash[25489]: ====================================================================
Oct 25 08:07:35 dotkat007 bash[25489]:       Set of internal maps for 'CMultiMetricMap' object
Oct 25 08:07:35 dotkat007 bash[25489]: ====================================================================
Oct 25 08:07:35 dotkat007 bash[25489]: Showing next the 1 internal maps:
Oct 25 08:07:35 dotkat007 bash[25489]: ------------------------- Internal map 1 out of 1 --------------------------
Oct 25 08:07:35 dotkat007 bash[25489]: -------------------------TMetricMapInitializer --------------------------
Oct 25 08:07:35 dotkat007 bash[25489]: ================ C++ Class: 'COccupancyGridMap2D'
Oct 25 08:07:35 dotkat007 bash[25489]: enableSaveAs3DObject                     = YES
Oct 25 08:07:35 dotkat007 bash[25489]: enableObservationLikelihood              = YES
Oct 25 08:07:35 dotkat007 bash[25489]: enableObservationInsertion               = YES
Oct 25 08:07:35 dotkat007 bash[25489]: min_x                                    = -10.000000
Oct 25 08:07:35 dotkat007 bash[25489]: max_x                                    = 10.000000
Oct 25 08:07:35 dotkat007 bash[25489]: min_y                                    = -10.000000
Oct 25 08:07:35 dotkat007 bash[25489]: max_y                                    = 10.000000
Oct 25 08:07:35 dotkat007 bash[25489]: resolution                               = 0.060000
Oct 25 08:07:35 dotkat007 bash[25489]: ----------- [COccupancyGridMap2D::TInsertionOptions] ------------
Oct 25 08:07:35 dotkat007 bash[25489]: mapAltitude                              = 0terminate called after throwing an instance of 'std::logic_error'
Oct 25 08:07:35 dotkat007 bash[25489]:   what():
Oct 25 08:07:35 dotkat007 bash[25489]:  =============== MRPT EXCEPTION =============
Oct 25 08:07:35 dotkat007 bash[25489]: void mrpt::utils::CStream::internal_ReadObject(mrpt::utils::CSerializablePtr&, mrpt::utils::CSerializable*) [with bool EXISTING_OBJ = true], line 428:
Oct 25 08:07:35 dotkat007 bash[25489]: Stored object has class 'mrpt::maps::COccupancyGridMap2D' which is not registered!
Oct 25 08:07:35 dotkat007 bash[25489]: Exception while parsing typed object 'mrpt::maps::COccupancyGridMap2D' from stream!
`

jlblancoc, 2022-11-27:
What example or launch file were you launching? What version of mrpt and/or mrpt_* packages (built from sources or from `apt install`?

Gunz-katomaran, 2022-11-28:
Thanks for your reply and contribution towards open source community.

MRPT version:
`ros-melodic-mrpt1:
  Installed: 1.5.9-1bionic.20201017.230653
  Candidate: 1.5.9-1bionic.20220514.021711
  Version table:
     1.5.9-1bionic.20220514.021711 500
        500 http://packages.ros.org/ros/ubuntu bionic/main amd64 Packages
 *** 1.5.9-1bionic.20201017.230653 100
        100 /var/lib/dpkg/status`
Picture for reference:
[image]

jlblancoc, 2022-11-28:
Mmm... you are using a **really old** version of mrpt, even for melodic.  mrpt1 is deprecated, please try this to upgrade: 

```bash
sudo apt update
sudo apt remove *mrpt*
sudo apt install ros-$ROS_DISTRO-mrpt-navigation
```

or if you want, for some reason, build it from sources, install mrpt deps like: 

```bash
sudo apt update
sudo apt remove *mrpt*
sudo apt install ros-$ROS_DISTRO-mrpt2
```

Gunz-katomaran, 2022-11-28:
Got the point. yeah, I will follow your instruction and provide the result @jlblancoc.  Once again thanks

Gunz-katomaran, 2022-12-09:
Thanks @jlblancoc , new version of MRPT solved the issue.

closed 2022-12-09 by Gunz-katomaran