mrpt-ros-pkg/mrpt_navigation #135: ROS 2 rework roadmap

jlblancoc, 2023-08-13:
List of goals for the ROS 2 transition. It will not only be a port of old ROS 1 packages, but a refactoring with the learned experience: 

- `mrpt_local_obstacles`: Sensor inputs to basic / decimated point cloud.
  - [ ] Port to ROS 2
  - [ ] Write tests

- `mrpt_msgs_bridge`: C++ library to convert msgs to C++ classes, for those messages defined in `mrpt_msgs` and hence not possible to include into `mrpt::ros2bridge`:
  - [ ] Port to ROS 2
  - [ ] Write tests

- `mrpt_map`: Read maps from file and publishes them.
  - [ ] Port existing ROS1 version to ROS 2
  - [ ] Write tests
  - [ ] Improve so it can read other map types: pointclouds, octomaps, etc.

- Old ROS 1 `mrpt_localization` should be renamed ==> `mrpt_pf_localization`: Uses particle filter methods to localize
  - [ ] Port old ROS 1 version to ROS 2
  - [ ] Rename package in all places where it's used: `mrpt_localization` ==> `mrpt_pf_localization`
  - [ ] Write tests
  - [ ] Expose and document the different options of particle filter algorithms that exist.
  - Write demos of localization using as map:
    - [ ] Occupancy grid map.
    - [ ] Point cloud.

- Create new package `mrpt_opt_localization`: Uses optimization-based localization:
  - Write new node. It should:
    - [ ] Optimize the pose by maximizing the observation likelihood of one or more sensors (2D and 3D lidars, RGBD, etc.) against a 2D or 3D grid or pointcloud.
    - [ ] Optionally: Fuse with wheels odometry
    - [ ] Optionally: fuse with IMU
    - [ ] Optionally: fuse with GNNS
  - [ ] Write tests
  - Write demos of localization using as map:
    - [ ] Occupancy grid map.
    - [ ] Point cloud.

- `mrpt_rawlog`: Nodes to record .rawlog and play .rawlog.
  - [ ] Port existing ROS1 version to ROS 2. Probably reuse code from mrpt/rosbag2rawlog app.
  - [ ] Write tests

- `mrpt_reactivenav2d`: The pure reactive method with a 2D robot shape for planar motion.
  - [ ] Port existing ROS1 version to ROS 2. Probably reuse code from mrpt/rosbag2rawlog app.
  - [ ] Write tests/demos

- `mrpt_tutorials`: Probably, rename to ==> `mrpt_navigation_tutorials`
  - [ ] Port existing ROS1 version to ROS 2.

Other tasks:
- [ ] Add proper CI and branch rules.
- [ ] When all/most of the above are done, switch the main branch of this repo to `ros2` and mark `ros1` as obsolete.
- [ ] Make the first ROS 2 release to all active distros.

jlblancoc, 2023-08-13:
cc: @SRai22 Here we are... :-)

SRai22, 2023-08-23:
Check and document what sensor types are supported. Check if https://github.com/mrpt-ros-pkg/mrpt_navigation/issues/121 is already handled. 

Currently we are handling both inputs, sensor_msgs::msg::LaserScan and  sensor_msgs::msg::PointCloud2 in mrpt_local_obstacles.

KatoROSTeam, 2023-11-03:
When can we Expect the mrpt_pf_localization

jlblancoc, 2023-11-06:
It's for sure one of my top priorities... but I'm almost done finishing first a brand new Lidar-SLAM package that will generate maps for these localization packages, so this one is enqueued at present... 

I definitively want both works to be out by December...

jlblancoc, 2023-12-20:
Finally back to this project...
These updated READMEs explain the new way metric maps are to be handled: 
- https://github.com/mrpt-ros-pkg/mrpt_navigation/tree/ros2/mrpt_map
- https://github.com/mrpt-ros-pkg/mrpt_navigation/tree/ros2/mrpt_pf_localization

(More news soon...)

KatoROSTeam, 2023-12-21:
Great to see you back here! Hoping mrpt_pf_localization will release on this December. @jlblancoc

jlblancoc, 2023-12-29:
Hi @KatoROSTeam : 
Officially releasing so users can install it via "apt install..." will take some more weeks, since there is a long queue of pending tasks to publish all the related packages (mola_common, mp2p_icp, etc.) and the ROS infrastructure guys are not moving things forwards these days (they also have the right to rest in holidays!). 

But you can test it if built locally. Basic pf-localization, with a 2D gridmap and a 2D lidar, is already working ok for ROS2 and there is a demo ready to test: https://github.com/mrpt-ros-pkg/mrpt_navigation/tree/ros2/mrpt_pf_localization#demos
with a video of the result here: https://mrpt.github.io/videos/pf_localization_demo_2dgrid_mvsim.mp4

To build locally, prepare a ROS2 workspace as usual, and clone in its `src`:
- https://github.com/MOLAorg/mola_common
- https://github.com/MOLAorg/mp2p_icp
- The latest MRPT2 (>=2.11.5) , must be also built from sources, unless you have ROS iron, since it's already published there: https://github.com/mrpt-ros-pkg/ros-build-farm-statuses
- Then, this repo, of course: https://github.com/mrpt-ros-pkg/mrpt_navigation/
- MVSim is required for the demo, but you can install it with `sudo apt install ros-${ROS_DISTRO}-mvsim` 

Hope it helps.

KatoROSTeam, 2024-01-20:
Hi @jlblancoc , In mrpt_pf_localization_node.cpp, i can find that lot of functions that required to localize on a given map are stopped from running using the #if 0 like below example,
```void PFLocalizationNode::odometryForCallback(
	CObservationOdometry::Ptr& _odometry,
	const std_msgs::msg::Header& _msg_header)
{
#if 0
	std::string base_frame_id = param()->base_frame_id;
	std::string odom_frame_id = param()->odom_frame_id;
	mrpt::poses::CPose3D poseOdom;
	if (this->waitForTransform(
			poseOdom, odom_frame_id, base_frame_id, _msg_header.stamp,
			ros::Duration(1.0)))
	{
		_odometry = CObservationOdometry::Create();
		_odometry->sensorLabel = odom_frame_id;
		_odometry->hasEncodersInfo = false;
		_odometry->hasVelocities = false;
		_odometry->odometry.x() = poseOdom.x();
		_odometry->odometry.y() = poseOdom.y();
		_odometry->odometry.phi() = poseOdom.yaw();
	}
#endif
}
```

Also the localization launch also not providing map frame as well. How to fix this

jlblancoc, 2024-03-07:
@KatoROSTeam  This package is now working on a daily basis on our robots, so you could use it now. 

@SRai22 I started importing our TPS A* planner into a new branch, and will try to port it to ROS 2 asap ;-)

logesh-gtx, 2025-02-11:
Will there be a localization package based on Landmark Maps in Mrpt ? (Landmarks -- such as QR codes )
or is there a way to generate landmark maps with Mrpt and how to use it for SLAM / Localization

jlblancoc, 2025-02-13:
@logesh-gtx It has been a long pending to-do item... 

MRPT C++ libraries do implement EKF-SLAM for 3D motion and 3D landmarks. It was exposed as ROS1 package [mrpt_slam](https://github.com/mrpt-ros-pkg/mrpt_slam), but it was never ported to ROS2 for lack of bandwidth from my side and lack of volunteers :-)

logesh-gtx, 2025-02-25:
I have thinking about integrating reflector Tags as landmarks in mrpt slam, but i dont have a concrete base on where i should start any suggestions or any documents available on MRPT? slam engine and how does it work with ros2

jlblancoc, 2025-03-03:
One way would be to start: 
- Copying this directory [mrpt_ekf_slam_3d](https://github.com/mrpt-ros-pkg/mrpt_slam/tree/ros1/mrpt_ekf_slam_3d) into a brand new repo, eg. named `mrpt_ekf_slam_3d`, now with the aim of being the new ros2 repo.
- Upgrade all the ROS1-related stuff to ROS2. ChatGPT is a good help here...
- Once done, the visual tags/markers should be converted (using its own independent package) into [mrpt_msgs/ObservationRangeBearing](https://github.com/mrpt-ros-pkg/mrpt_msgs/blob/master/msg-ros1/ObservationRangeBearing.msg). That message is already defined and working in ROS1 & ROS2 in the package `mrpt_msgs`

jlblancoc, 2025-03-03:
PS: I realized you never actually asked for ROS2... if you still work in ROS1, the only missing piece is the third bullet above.