mrpt-ros-pkg/mrpt_navigation #138: Drop dependency on marker_msgs

jlblancoc, 2023-08-18:
Build error on the build farm:

    KeyError: 'marker_msgs'

Reason: that package is not released anymore.

Solution: drop the use of that package and replace by any other standard way to send markers to rviz (I think that was it's use). MVSim does that already, for reference.

maxbader, 2023-08-19:
marker_msgs is allready released for humble

jlblancoc, 2023-08-19:
Hi @maxbader ! 

Ah, I thought it was not. I'll consider the different options... 

Thanks.

jlblancoc, 2023-08-19:
PS: as an advice: always release first to rolling. That way, new distros will always already come with a version of your package...

closed 2023-08-19 by jlblancoc

jlblancoc, 2023-08-19:
@maxbader Ok, I reviewed the code and the only relevant use was in mrpt_msgs_bridge, so I made that code build-dependent to whether marker_msgs is present in the system or not, and commented it out from package.xml. 

Once marker_msgs is available in humble, iron, rolling, feel free of editing/pull-request to uncomment that line from package.xml and it will be included and working again!

commit 0c44d9b1 (2023-08-19): Make marker_msgs an optional dependency.

Closes #138