mrpt-ros-pkg/mrpt_navigation #140: Migrate to modern Gazebo

azeey, 2023-12-21:
As you may know, Gazebo-classic (aka Gazebo11, see [Terminology](https://gazebosim.org/about#:~:text=Terminology)) is no longer being actively developed  and will not be available on ROS Jazzy ([REP 2000](https://www.ros.org/reps/rep-2000.html#jazzy-jalisco-may-2024-may-2029)). I noticed that this project still uses Gazebo-classic. Are there any plans to migrate to modern Gazebo ([previously known as Ignition](https://community.gazebosim.org/t/a-new-era-for-gazebo/1356))?

jlblancoc, 2024-01-03:
Thanks for the pointers, @azeey ! 
Indeed, we'll probably update those simulation-based files before the next official ROS 2 release.

azeey, 2024-01-05:
Great! Thanks for the update.