mrpt-ros-pkg/mrpt_navigation #150: [Noetic] mrpt_local_obstacles fails to build on buildfarm

sloretz, 2024-09-16:
https://build.ros.org/view/Nbin_uF64/job/Nbin_uF64__mrpt_local_obstacles__ubuntu_focal_amd64__binary/99/console

Looks like a few new compile errors

```
12:37:25 /tmp/binarydeb/ros-noetic-mrpt-local-obstacles-1.0.6/submodules/mp2p_icp/mp2p_icp/include/mp2p_icp/Matcher_Points_DistanceThreshold.h:51:10: error: ‘void mp2p_icp::Matcher_Points_DistanceThreshold::initialize(const mrpt::containers::yaml&)’ marked ‘override’, but does not override
12:37:25    51 |     void initialize(const mrpt::containers::yaml& params) override;
12:37:25       |          ^~~~~~~~~~
12:37:25 In file included from /tmp/binarydeb/ros-noetic-mrpt-local-obstacles-1.0.6/submodules/mp2p_icp/mp2p_icp/include/mp2p_icp/ICP.h:19,
12:37:25                  from /tmp/binarydeb/ros-noetic-mrpt-local-obstacles-1.0.6/submodules/mp2p_icp/mp2p_icp/src/ICP.cpp:13:
12:37:25 /tmp/binarydeb/ros-noetic-mrpt-local-obstacles-1.0.6/submodules/mp2p_icp/mp2p_icp/include/mp2p_icp/QualityEvaluator_PairedRatio.h:34:10: error: ‘void mp2p_icp::QualityEvaluator_PairedRatio::initialize(const mrpt::containers::yaml&)’ marked ‘override’, but does not override
12:37:25    34 |     void initialize(const mrpt::containers::yaml& params) override;
12:37:25       |          ^~~~~~~~~~
12:37:25 In file included from /tmp/binarydeb/ros-noetic-mrpt-local-obstacles-1.0.6/submodules/mp2p_icp/mp2p_icp/include/mp2p_icp/ICP.h:21,
12:37:25                  from /tmp/binarydeb/ros-noetic-mrpt-local-obstacles-1.0.6/submodules/mp2p_icp/mp2p_icp/src/ICP.cpp:13:
12:37:25 /tmp/binarydeb/ros-noetic-mrpt-local-obstacles-1.0.6/submodules/mp2p_icp/mp2p_icp/include/mp2p_icp/Solver.h:48:5: error: ‘DEFINE_VIRTUAL_MRPT_OBJECT’ does not name a type
12:37:25    48 |     DEFINE_VIRTUAL_MRPT_OBJECT(Solver)
12:37:25       |     ^~~~~~~~~~~~~~~~~~~~~~~~~~
12:37:25 /tmp/binarydeb/ros-noetic-mrpt-local-obstacles-1.0.6/submodules/mp2p_icp/mp2p_icp/src/ICP.cpp: In static member function ‘static bool mp2p_icp::ICP::run_solvers(const solver_list_t&, const mp2p_icp::Pairings&, mp2p_icp::OptimalTF_Result&, const mp2p_icp::SolverContext&)’:
12:37:26 /tmp/binarydeb/ros-noetic-mrpt-local-obstacles-1.0.6/submodules/mp2p_icp/mp2p_icp/src/ICP.cpp:253:21: error: ‘using element_type = class mrpt::rtti::CObject’ {aka ‘class mrpt::rtti::CObject’} has no member named ‘optimal_pose’
12:37:26   253 |         if (solver->optimal_pose(pairings, out, sc)) return true;
12:37:26       |                     ^~~~~~~~~~~~
12:37:26 /tmp/binarydeb/ros-noetic-mrpt-local-obstacles-1.0.6/submodules/mp2p_icp/mp2p_icp/src/ICP.cpp: In static member function ‘static void mp2p_icp::ICP::initialize_solvers(const mrpt::containers::yaml&, mp2p_icp::ICP::solver_list_t&)’:
12:37:26 /tmp/binarydeb/ros-noetic-mrpt-local-obstacles-1.0.6/submodules/mp2p_icp/mp2p_icp/src/ICP.cpp:285:12: error: ‘using element_type = class mp2p_icp::Solver’ {aka ‘class mp2p_icp::Solver’} has no member named ‘initialize’
12:37:26   285 |         m->initialize(e.at("params"));
12:37:26       |            ^~~~~~~~~~
12:37:26 /tmp/binarydeb/ros-noetic-mrpt-local-obstacles-1.0.6/submodules/mp2p_icp/mp2p_icp/src/ICP.cpp: In static member function ‘static void mp2p_icp::ICP::initialize_matchers(const mrpt::containers::yaml&, mp2p_icp::matcher_list_t&)’:
12:37:26 /tmp/binarydeb/ros-noetic-mrpt-local-obstacles-1.0.6/submodules/mp2p_icp/mp2p_icp/src/ICP.cpp:317:12: error: ‘using element_type = class mp2p_icp::Matcher’ {aka ‘class mp2p_icp::Matcher’} has no member named ‘initialize’
12:37:26   317 |         m->initialize(e.at("params"));
12:37:26       |            ^~~~~~~~~~
12:37:26 /tmp/binarydeb/ros-noetic-mrpt-local-obstacles-1.0.6/submodules/mp2p_icp/mp2p_icp/src/ICP.cpp: In static member function ‘static void mp2p_icp::ICP::initialize_quality_evaluators(const mrpt::containers::yaml&, mp2p_icp::ICP::quality_eval_list_t&)’:
12:37:26 /tmp/binarydeb/ros-noetic-mrpt-local-obstacles-1.0.6/submodules/mp2p_icp/mp2p_icp/src/ICP.cpp:346:12: error: ‘using element_type = class mp2p_icp::QualityEvaluator’ {aka ‘class mp2p_icp::QualityEvaluator’} has no member named ‘initialize’
12:37:26   346 |         m->initialize(e.at("params"));
12:37:26       |            ^~~~~~~~~~
12:37:26 /tmp/binarydeb/ros-noetic-mrpt-local-obstacles-1.0.6/submodules/mp2p_icp/mp2p_icp/src/ICP.cpp: In static member function ‘static double mp2p_icp::ICP::evaluate_quality(const quality_eval_list_t&, const mp2p_icp::metric_map_t&, const mp2p_icp::metric_map_t&, const mrpt::poses::CPose3D&, const mp2p_icp::Pairings&)’:
12:37:26 /tmp/binarydeb/ros-noetic-mrpt-local-obstacles-1.0.6/submodules/mp2p_icp/mp2p_icp/src/ICP.cpp:373:20: error: ‘using element_type = class mrpt::rtti::CObject’ {aka ‘class mrpt::rtti::CObject’} has no member named ‘evaluate’
12:37:26   373 |             e.obj->evaluate(pcGlobal, pcLocal, localPose, finalPairings);
12:37:26       |                    ^~~~~~~~
12:37:26 make[4]: *** [submodules/mp2p_icp/mp2p_icp/CMakeFiles/mp2p_icp.dir/build.make:66: submodules/mp2p_icp/mp2p_icp/CMakeFiles/mp2p_icp.dir/src/ICP.cpp.o] Error 1
```

jlblancoc, 2024-09-16:
Thanks!
Let me check first if https://github.com/ros/rosdistro/pull/42882 finally fixes mp2p_icp builds in Noetic as a proper independent package (it [fails since ~1 mo ago](https://build.ros.org/job/Nbin_uF64__mp2p_icp__ubuntu_focal_amd64__binary/)), then I would update this `mrpt_navigation` ROS1 branch to use mp2p_icp as a `<depend>` instead of a submodule, fixed to an old commit version (sigh).

jlblancoc, 2024-09-16:
@sloretz  Thanks for the quick merge... it worked!

This repo is probably fixed after 42bfb34d57ac9912e14e26e00188b90141ad8ee2
Let's wait for the next Ndev job and if it's green I'll make a new bloom-release.

jlblancoc, 2024-09-17:
@sloretz It worked. This PR should fix the regression https://github.com/ros/rosdistro/pull/42883

closed 2024-09-18 by jlblancoc