mrpt-ros-pkg/mrpt_navigation #151: MRPT PF localization gets killed at random starts

logesh-gtx, 2024-09-27:
hello, 
I'm currently using the mrpt-navigation 2.1.1 ---> installed using `sudo apt install ros-humble-mrpt-navigation` 
ROS version --> humble running in ubuntu 22.04

at random launches of the mrpt pf-localization, I get this below mentioned error

[image]

This error shows that it waits for lidar transform frame to be available but gets killed after that, 
I'm currently publishing the base->lidar transform using static TF, but at some instances it gets killed in others it running fine as expected, 
Is anything is possible to avoid this issue

jlblancoc, 2024-09-27:
Hi, 
Yes, that's a known issue that happens when /tf for sensor poses are not received "in time", or do not exist. From your "at random" I guess it does work sometimes, so your /tf exist and it's just a problem of timing. 

The 2.2.0 version fixed this (see [changelog](https://github.com/mrpt-ros-pkg/mrpt_navigation/blob/ros2/mrpt_pf_localization/CHANGELOG.rst#220-2024-09-25)). 
There will probably be no Humble sync for 2-3 weeks, so you can uninstall this package with apt and clone and build the latest version from sources to get rid of the issue.

By the way: the implemented fix is that those sensor scans with missing /tf are just ignored, and normally the next one works fine. In this way, it always works for us. 
PS: Any other idea on how to deal with /tf timing issues would be welcome!

logesh-gtx, 2024-10-01:
ok thank you @jlblancoc i'll use the specified version

closed 2024-10-01 by logesh-gtx