mrpt-ros-pkg/mrpt_navigation #154: Cannot play rawlog file with rawlog_play_node

JeffR1992, 2025-01-09:
I'm using ROS1 Noetic on Ubuntu 20.04 and am trying to play the Malaga 2006 campus rawlog dataset (https://www.mrpt.org/Malaga_2006_campus_dataset).

I've successfully built `mrpt_navigation` in my `catkin_ws` using the `ros1` branch (`tag: 1.0.7, origin/ros1`). However, when I try and play the Malaga 2006 dataset using:
`rosrun mrpt_rawlog rawlog_play_node 20060121-Teleco_Faculty_Malaga_laser_only.rawlog`

I get the following error output:
```
[ INFO] [1736402505.739311969]: rate: 10.000000
[ INFO] [1736402505.740450533]: parameter_update_skip: 1
[ INFO] [1736402505.740662877]: rawlog_file: data.rawlog
[ INFO] [1736402505.741303444]: odom_frame: odom
[ INFO] [1736402505.741582365]: base_frame: base_link
[ INFO] [1736402505.741622481]: gaussianModel.a1: 0.034000
[ INFO] [1736402505.741661545]: gaussianModel.a2: 0.057000
[ INFO] [1736402505.741669891]: gaussianModel.a3: 0.014000
[ INFO] [1736402505.741675712]: gaussianModel.a4: 0.097000
[ INFO] [1736402505.741680251]: gaussianModel.minStdXY: 0.005000
[ INFO] [1736402505.741689067]: gaussianModel.minStdPHI: 0.050000
[ERROR] [1736402505.744896504]: raw_file: data.rawlog does not exit
```

I have tried using `raw_play_node` on other datasets, such as Edmonton 2002 (https://www.mrpt.org/Dataset_Edmonton_2002), but I get a similar error. Is there something that I'm doing wrong when calling `raw_play_node`? Any help would be appreciated.

jlblancoc, 2025-01-09:
Hi, 
The issue here is: 

    [ INFO] [1736402505.740662877]: rawlog_file: data.rawlog

so, the ROS **parameter** "rawlog_file" is not set. It's been a long time since I don't use this ROS1 version but it should be like: 

    rosrun mrpt_rawlog rawlog_play_node _rawlog_file:=/full/path/to/20060121-Teleco_Faculty_Malaga_laser_only.rawlog

If you move at some point to ROS2, the most up-to-date way to publish rawlog files (and many other dataset formats) to ROS2 is via MOLA (see [this page](https://docs.mola-slam.org/latest/dataset-conversions.html)).

JeffR1992, 2025-01-10:
Thanks @jlblancoc, that solved it.

closed 2025-01-10 by JeffR1992